supermerill/SuperSlicer #703: fail build in archlinux

sl1pkn07, 2020-11-22:
### Version
0fe20a6dc460154af16eb531953c048147ee001c

### Operating system type + version
Archlinux

### 3D printer brand / version + firmware version (if known)
Ender 3 + Marlin bugfix-2.0.x

### Behavior

https://github.com/supermerill/SuperSlicer/commit/05632802c18a3ec560f8b43575b2fa0f42cef008#commitcomment-44428911
https://github.com/supermerill/SuperSlicer/commit/cae61c82ad78df00169c256f100bb3b9307f93e5#commitcomment-44429971

and
~~~make
[ 26%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/PrintConfig.cpp.o
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/PrintConfig.cpp: En la función miembro static ‘static void Slic3r::PrintConfigDef::to_prusa(Slic3r::t_config_option_key&, std::string&, const Slic3r::DynamicConfig&)’:
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/PrintConfig.cpp:4948:10: error: ‘unordered_set’ no es un miembro de ‘std’
 4948 |     std::unordered_set<std::string> to_remove_keys = {
      |          ^~~~~~~~~~~~~
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/PrintConfig.cpp:12:1: nota: ‘std::unordered_set’ is defined in header ‘<unordered_set>’; did you forget to ‘#include <unordered_set>’?
   11 | #include <boost/nowide/iostream.hpp>
  +++ |+#include <unordered_set>
   12 | 
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/PrintConfig.cpp:4948:35: error: expected primary-expression before ‘>’ token
 4948 |     std::unordered_set<std::string> to_remove_keys = {
      |                                   ^
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/PrintConfig.cpp:4948:37: error: ‘to_remove_keys’ no se declaró en este ámbito
 4948 |     std::unordered_set<std::string> to_remove_keys = {
      |                                     ^~~~~~~~~~~~~~
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/build.make:992: src/libslic3r/CMakeFiles/libslic3r.dir/PrintConfig.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1185: src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
~~~

fixed with:

~~~patch
diff --git a/src/libslic3r/GCode/FanMover.cpp b/src/libslic3r/GCode/FanMover.cpp
index 8ce7d7d2c..3b8aebaac 100644
--- a/src/libslic3r/GCode/FanMover.cpp
+++ b/src/libslic3r/GCode/FanMover.cpp
@@ -2,6 +2,8 @@

 #include "GCodeReader.hpp"

+#include <iomanip>
+
 /*
 #include <memory.h>
 #include <string.h>
diff --git a/src/libslic3r/GCode/GCodeProcessor.cpp b/src/libslic3r/GCode/GCodeProcessor.cpp
index a420d4404..1cc8e4f20 100644
--- a/src/libslic3r/GCode/GCodeProcessor.cpp
+++ b/src/libslic3r/GCode/GCodeProcessor.cpp
@@ -6,6 +6,7 @@
 #include <boost/log/trivial.hpp>
 #include <boost/nowide/fstream.hpp>
 #include <boost/nowide/cstdio.hpp>
+#include <boost/algorithm/string/case_conv.hpp>

 #include <float.h>
 #include <assert.h>
diff --git a/src/libslic3r/PrintConfig.cpp b/src/libslic3r/PrintConfig.cpp
index bca4793e3..d708e54c8 100644
--- a/src/libslic3r/PrintConfig.cpp
+++ b/src/libslic3r/PrintConfig.cpp
@@ -9,6 +9,7 @@
 #include <boost/lexical_cast.hpp>
 #include <boost/thread.hpp>
 #include <boost/nowide/iostream.hpp>
+#include <unordered_set>

 #include <float.h>
~~~

but now:

~~~make
[ 25%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/VoronoiOffset.cpp.o
In file included from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/VoronoiOffset.cpp:3:
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/VoronoiOffset.hpp:21:26: error: 'segment_data' in namespace 'boost::polygon' does not name a template type; did you mean 'segment_traits'?
   21 |  typedef boost::polygon::segment_data<coordinate_type>   segment_type;
      |                          ^~~~~~~~~~~~
      |                          segment_traits
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/VoronoiOffset.hpp:31:1: error: 'Polygons' does not name a type
   31 | Polygons voronoi_offset(
      | ^~~~~~~~
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/build.make:1278: src/libslic3r/CMakeFiles/libslic3r.dir/VoronoiOffset.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1185: src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
~~~

greetings

supermerill, 2020-11-22:
`#include "boost/polygon/segment_data.hpp` in VoronoiOffset.hpp?

sl1pkn07, 2020-11-22:
yep. fixed with

~~~patch
diff --git a/src/libslic3r/VoronoiOffset.hpp b/src/libslic3r/VoronoiOffset.hpp
index 423ec68f1..389050c0b 100644
--- a/src/libslic3r/VoronoiOffset.hpp
+++ b/src/libslic3r/VoronoiOffset.hpp
@@ -7,6 +7,8 @@

 #define BOOST_VORONOI_USE_GMP 1
 #include "boost/polygon/voronoi.hpp"
+#include "boost/polygon/segment_data.hpp"
+#include "Polygon.hpp"
 using boost::polygon::voronoi_builder;
 using boost::polygon::voronoi_diagram;
 
~~~

but now:

~~~make
[ 61%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/CalibrationBridgeDialog.cpp.o
In file included from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/PrintBase.hpp:14,
                 from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Print.hpp:4,
                 from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/Jobs/ArrangeJob.hpp:6,
                 from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/CalibrationBridgeDialog.cpp:6:
/usr/include/tbb/mutex.h:21:140: note: '#pragma message: TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.'
   21 | #pragma message("TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                                                                                                                                            ^
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/CalibrationBridgeDialog.cpp: In member function 'void Slic3r::GUI::CalibrationBridgeDialog::create_geometry(std::string, bool)':
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/CalibrationBridgeDialog.cpp:110:23: warning: comparison of integer expressions of different signedness: 'int' and 'size_t' {aka 'long unsigned int'} [-Wsign-compare]
  110 |     for (int i = 1; i < nb_items; i++) {
      |                     ~~^~~~~~~~~~
In file included from /usr/include/c++/10.2.0/memory:84,
                 from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/GUI_App.hpp:4,
                 from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/CalibrationAbstractDialog.hpp:9,
                 from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/CalibrationBridgeDialog.hpp:4,
                 from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/CalibrationBridgeDialog.cpp:1:
/usr/include/c++/10.2.0/bits/shared_ptr.h: In instantiation of 'std::shared_ptr<_Tp> std::dynamic_pointer_cast(const std::shared_ptr<_Tp>&) [with _Tp = Slic3r::ProgressIndicator; _Up = Slic3r::GUI::ProgressIndicatorStub]':
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/CalibrationBridgeDialog.cpp:154:88:   required from here
/usr/include/c++/10.2.0/bits/shared_ptr.h:602:23: error: 'Slic3r::ProgressIndicator' is an inaccessible base of 'Slic3r::GUI::ProgressIndicatorStub'
  602 |       if (auto* __p = dynamic_cast<typename _Sp::element_type*>(__r.get()))
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/10.2.0/bits/shared_ptr.h:602:23: error: 'Slic3r::ProgressIndicator' is an inaccessible base of 'Slic3r::GUI::ProgressIndicatorStub'
make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/build.make:186: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/CalibrationBridgeDialog.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1307: src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
~~~

sl1pkn07, 2020-11-22:
the error posted avobe is gone (i'm not sure why), but now get:

~~~make
[ 61%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/FreeCADDialog.cpp.o
In file included from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp:1:
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/FreeCADDialog.hpp:79:5: error: 'wxComboBox' does not name a type
   79 |     wxComboBox* cmb_add_replace;
      |     ^~~~~~~~~~
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp: In constructor 'Slic3r::GUI::FreeCADDialog::FreeCADDialog(Slic3r::GUI::GUI_App*, Slic3r::GUI::MainFrame*)':
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp:287:5: error: 'cmb_add_replace' was not declared in this scope
  287 |     cmb_add_replace = new wxComboBox(this, wxID_ANY, wxString{ "replace" }, wxDefaultPosition, wxDefaultSize, 2, choices_add);
      |     ^~~~~~~~~~~~~~~
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp: In member function 'bool Slic3r::GUI::FreeCADDialog::load_text_from_file(const boost::filesystem::path&)':
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp:369:31: warning: catching polymorphic type 'class std::exception' by value [-Wcatch-value=]
  369 |         catch (std::exception ex) {
      |                               ^~
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp: In member function 'bool Slic3r::GUI::FreeCADDialog::write_text_in_file(const wxString&, const boost::filesystem::path&)':
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp:410:27: warning: catching polymorphic type 'class std::exception' by value [-Wcatch-value=]
  410 |     catch (std::exception ex) {
      |                           ^~
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp: In member function 'void Slic3r::GUI::FreeCADDialog::test_update_script_file(std::string&)':
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp:833:27: warning: catching polymorphic type 'class std::exception' by value [-Wcatch-value=]
  833 |     catch (std::exception ex) {
      |                           ^~
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp: In member function 'void Slic3r::GUI::FreeCADDialog::create_geometry(wxCommandEvent&)':
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp:1028:8: error: 'cmb_add_replace' was not declared in this scope
 1028 |     if(cmb_add_replace->GetSelection() == 0)
      |        ^~~~~~~~~~~~~~~
make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/build.make:290: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/FreeCADDialog.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1307: src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
~~~

for reference, this is my cmake config

~~~
  cmake ../SuperSlicer \
    -DCMAKE_BUILD_TYPE=None \
    -DCMAKE_INSTALL_PREFIX=/usr \
    -DOpenGL_GL_PREFERENCE=GLVND \
    -DSLIC3R_FHS=ON \
    -DSLIC3R_PCH=OFF \
    -DSLIC3R_STATIC=OFF \
    -DSLIC3R_WX_STABLE=OFF \
    -DSLIC3R_GTK=3 \
    -DSLIC3R_BUILD_TESTS=OFF \
    -DwxWidgets_CONFIG_EXECUTABLE=/usr/bin/wx-config-3.1 \
    -DwxWidgets_wxrc_EXECUTABLE=/usr/bin/wxrc-3.1 \
    -DBoost_NO_BOOST_CMAKE=TRUE \
    -DBoost_NO_SYSTEM_PATHS=TRUE  \
    -DBOOST_ROOT=/opt/usr \
    -DBOOST_INCLUDEDIR=/opt/usr/include \
    -DBOOST_LIBRARYDIR=/opt/usr/lib
~~~

wxwidgets 3.1.4
boot 1.71.0 (due error with 1.72.0)
gcc 10.2.0

supermerill, 2020-11-22:
add `#include <wx/combobox.h>` ~l22 in SuperSlicer/src/slic3r/GUI/FreeCADDialog.cpp

sl1pkn07, 2020-11-23:
hi

in that file no. but in SuperSlicer/src/slic3r/GUI/FreeCADDialog.hpp yes

but also spotted other errors, but fixed (in my kwedge) with

~~~patch
diff --git a/src/slic3r/GUI/FreeCADDialog.hpp b/src/slic3r/GUI/FreeCADDialog.hpp
index 22a864e8b..1f8a44462 100644
--- a/src/slic3r/GUI/FreeCADDialog.hpp
+++ b/src/slic3r/GUI/FreeCADDialog.hpp
@@ -9,6 +9,7 @@
 
 #include <wx/gbsizer.h>
 #include <wx/stc/stc.h>
+#include <wx/combobox.h>
 
 namespace Slic3r { 
 namespace GUI {
diff --git a/src/slic3r/GUI/GCodeViewer.cpp b/src/slic3r/GUI/GCodeViewer.cpp
index 9d1363770..bb2378ba8 100644
--- a/src/slic3r/GUI/GCodeViewer.cpp
+++ b/src/slic3r/GUI/GCodeViewer.cpp
@@ -23,6 +23,11 @@
 #include <GL/glew.h>
 #include <boost/log/trivial.hpp>
 #include <boost/nowide/cstdio.hpp>
+#include <boost/nowide/fstream.hpp>
+#include <boost/property_tree/ptree.hpp>
+#include <boost/property_tree/ini_parser.hpp>
+#include <boost/locale/generator.hpp>
+
 #include <wx/progdlg.h>
 #include <wx/numformatter.h>
 
diff --git a/src/libslic3r/ExPolygon.cpp b/src/libslic3r/ExPolygon.cpp
index 503f7d1ba..4c86a42a9 100644
--- a/src/libslic3r/ExPolygon.cpp
+++ b/src/libslic3r/ExPolygon.cpp
@@ -1,7 +1,8 @@
 #include "BoundingBox.hpp"
 #include "ExPolygon.hpp"
 
-#include "Exception.hpp"#include "MedialAxis.hpp"
+#include "Exception.hpp"
+#include "MedialAxis.hpp"
 #include "Geometry.hpp"
 #include "Polygon.hpp"
 #include "Line.hpp"
~~~

but spotted other

~~~
[ 62%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/GUI_Preview.cpp.o
En el fichero incluido desde /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/PrintBase.hpp:14,
                 desde /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/BackgroundSlicingProcess.hpp:10,
                 desde /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/GUI_Preview.cpp:7:
/usr/include/tbb/mutex.h:21:140: nota: ‘#pragma message: TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.’
   21 | #pragma message("TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                                                                                                                                            ^
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/GUI_Preview.cpp: En la función miembro ‘bool Slic3r::GUI::Preview::init(wxWindow*, Slic3r::Model*)’:
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/GUI_Preview.cpp:220:25: error: ‘Get’ was not declared in this scope; did you mean ‘GetId’?
  220 |     SetBackgroundColour(Get*Parent()->GetBackgroundColour());
      |                         ^~~
      |                         GetId
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/GUI_Preview.cpp:220:29: error: ‘Parent’ was not declared in this scope; did you mean ‘parent’?
  220 |     SetBackgroundColour(Get*Parent()->GetBackgroundColour());
      |                             ^~~~~~
      |                             parent
make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/build.make:654: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/GUI_Preview.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1307: src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
~~~

greetings

supermerill, 2020-11-23:
delete the `*` between Get and Parent.
pushed on master

Lukas1818, 2020-11-23:
I tested the change  of adf6f72a1b0f99f82efe689f3f28bd74eae5caaf and it seems to fix the issue.

sl1pkn07, 2020-11-23:
Hi

not at all

with this

~~~patch
diff --git a/src/slic3r/Utils/PresetUpdater.cpp b/src/slic3r/Utils/PresetUpdater.cpp
index 1aec70634..e945c6cc1 100644
--- a/src/slic3r/Utils/PresetUpdater.cpp
+++ b/src/slic3r/Utils/PresetUpdater.cpp
@@ -12,6 +12,8 @@
 #include <boost/filesystem/fstream.hpp>
 #include <boost/lexical_cast.hpp>
 #include <boost/log/trivial.hpp>
+#include <boost/property_tree/ptree.hpp>
+#include <boost/property_tree/json_parser.hpp>
 
 #include <wx/app.h>
 #include <wx/msgdlg.h>
~~~

now i abre to build enterely

and with this, i can selence a warning when do cmake "configure"

~~~patch
diff --git a/cmake/modules/FindDBus.cmake b/cmake/modules/FindDBus.cmake
index 1d0f29dd7..d54d4e516 100644
--- a/cmake/modules/FindDBus.cmake
+++ b/cmake/modules/FindDBus.cmake
@@ -56,4 +56,4 @@ FIND_PATH(DBUS_ARCH_INCLUDE_DIR
 SET(DBUS_INCLUDE_DIRS ${DBUS_INCLUDE_DIR} ${DBUS_ARCH_INCLUDE_DIR})
 
 INCLUDE(FindPackageHandleStandardArgs)
-FIND_PACKAGE_HANDLE_STANDARD_ARGS(DBUS REQUIRED_VARS DBUS_INCLUDE_DIRS DBUS_LIBRARIES)
\ No newline at end of file
+FIND_PACKAGE_HANDLE_STANDARD_ARGS(DBus REQUIRED_VARS DBUS_INCLUDE_DIRS DBUS_LIBRARIES)
\ No newline at end of file
~~~

about warmning problem with the name of the cmake DBus finder

greetings

sl1pkn07, 2020-11-23:
and also i get this

~~~make
CMake Deprecation Warning at src/admesh/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at src/boost/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at src/clipper/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at src/miniz/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Miniz NOT found in system, using bundled version...
CMake Deprecation Warning at src/glu-libtess/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at src/polypartition/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at src/poly2tri/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Using qhull from system.
CMake Deprecation Warning at src/Shiny/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at src/semver/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at src/imgui/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- wx-config path: /usr/bin/wx-config-3.1
~~~

sl1pkn07, 2020-11-23:
unafortunate. the app freeze when slice any piece :/

Build OK
App crash

:(

sl1pkn07, 2020-12-12:
Hi again

when try to build with using tag `2.3.55.2` and boost 1.75.0

~~~make
[ 26%] Building CXX object src/libslic3r/CMakeFiles/libslic3r.dir/Thread.cpp.o
In file included from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Thread.cpp:12:
/usr/include/tbb/tbb_thread.h:21:145: note: '#pragma message: TBB Warning: tbb/tbb_thread.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.'
   21 | #pragma message("TBB Warning: tbb/tbb_thread.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                                                                                                                                                 ^
In file included from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Thread.cpp:13:
/usr/include/tbb/task_scheduler_init.h:21:154: note: '#pragma message: TBB Warning: tbb/task_scheduler_init.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.'
   21 | #pragma message("TBB Warning: tbb/task_scheduler_init.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                                                                                                                                                          ^
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Thread.cpp: In function 'void Slic3r::name_tbb_thread_pool_threads()':
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Thread.cpp:208:39: error: variable 'std::atomic<long unsigned int> nthreads_running' has initializer but incomplete type
  208 |  std::atomic<size_t>  nthreads_running(0);
      |                                       ^
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Thread.cpp: In lambda function:
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Thread.cpp:216:14: error: 'nthreads_running' is not captured
  216 |          if (nthreads_running.fetch_add(1) + 1 == nthreads) {
      |              ^~~~~~~~~~~~~~~~
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Thread.cpp:214:68: note: the lambda has no capture-default
  214 |         [&nthreads_running, nthreads, &master_thread_id, &cv, &cv_m](const tbb::blocked_range<size_t> &range) {
      |                                                                    ^
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Thread.cpp:208:23: note: '<typeprefixerror>nthreads_running' declared here
  208 |  std::atomic<size_t>  nthreads_running(0);
      |                       ^~~~~~~~~~~~~~~~
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Thread.cpp:223:22: error: 'nthreads_running' is not captured
  223 |        cv.wait(lk, [&nthreads_running, nthreads]{return nthreads_running == nthreads;});
      |                      ^~~~~~~~~~~~~~~~
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Thread.cpp:214:68: note: the lambda has no capture-default
  214 |         [&nthreads_running, nthreads, &master_thread_id, &cv, &cv_m](const tbb::blocked_range<size_t> &range) {
      |                                                                    ^
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Thread.cpp:208:23: note: '<typeprefixerror>nthreads_running' declared here
  208 |  std::atomic<size_t>  nthreads_running(0);
      |                       ^~~~~~~~~~~~~~~~
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Thread.cpp: In lambda function:
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Thread.cpp:223:57: error: 'nthreads_running' is not captured
  223 |        cv.wait(lk, [&nthreads_running, nthreads]{return nthreads_running == nthreads;});
      |                                                         ^~~~~~~~~~~~~~~~
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Thread.cpp:223:48: note: the lambda has no capture-default
  223 |        cv.wait(lk, [&nthreads_running, nthreads]{return nthreads_running == nthreads;});
      |                                                ^
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Thread.cpp:208:23: note: '<typeprefixerror>nthreads_running' declared here
  208 |  std::atomic<size_t>  nthreads_running(0);
      |                       ^~~~~~~~~~~~~~~~
make[2]: *** [src/libslic3r/CMakeFiles/libslic3r.dir/build.make:1239: src/libslic3r/CMakeFiles/libslic3r.dir/Thread.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1185: src/libslic3r/CMakeFiles/libslic3r.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
~~~

greetings

sl1pkn07, 2021-09-25:
~~~make
[ 63%] Building CXX object src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Jobs/ArrangeJob.cpp.o
In file included from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/PrintBase.hpp:14,
                 from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Print.hpp:4,
                 from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/Jobs/ArrangeJob.hpp:6,
                 from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/Jobs/ArrangeJob.cpp:1:
/usr/include/tbb/mutex.h:21:140: note: '#pragma message: TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.'
   21 | #pragma message("TBB Warning: tbb/mutex.h is deprecated. For details, please see Deprecated Features appendix in the TBB reference manual.")
      |                                                                                                                                            ^
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/Jobs/ArrangeJob.cpp: In function 'void Slic3r::GUI::add_brim(Slic3r::arrangement::ArrangePolygon&, const Slic3r::ModelConfigObject&, const Slic3r::GUI::Plater*)':
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/Jobs/ArrangeJob.cpp:81:34: error: 'offset_ex' was not declared in this scope; did you mean 'offsetof'?
   81 |             ExPolygons brimmed = offset_ex(ap.poly, diff);
      |                                  ^~~~~~~~~
      |                                  offsetof
In file included from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Flow.hpp:5,
                 from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/MedialAxis.hpp:9,
                 from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Geometry.hpp:6,
                 from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Model.hpp:5,
                 from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/PrintBase.hpp:17,
                 from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Print.hpp:4,
                 from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/Jobs/ArrangeJob.hpp:6,
                 from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/slic3r/GUI/Jobs/ArrangeJob.cpp:1:
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Config.hpp: In instantiation of 'static Slic3r::t_config_enum_names Slic3r::ConfigOptionEnum<T>::create_enum_names() [with T = Slic3r::SLAPillarConnectionMode; Slic3r::t_config_enum_names = std::vector<std::__cxx11::basic_string<char> >]':
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Config.hpp:1580:82:   required from 'std::string Slic3r::ConfigOptionEnum<T>::serialize() const [with T = Slic3r::SLAPillarConnectionMode; std::string = std::__cxx11::basic_string<char>]'
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Config.hpp:1577:17:   required from here
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Config.hpp:1607:57: warning: loop variable 'kvp' of type 'const std::pair<std::__cxx11::basic_string<char>, int>&' binds to a temporary constructed from type 'const std::pair<const std::__cxx11::basic_string<char>, int>' [-Wrange-loop-construct]
 1607 |             for (const std::pair<std::string, int32_t> &kvp : enum_keys_map)
      |                                                         ^~~
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Config.hpp:1607:57: note: use non-reference type 'const std::pair<std::__cxx11::basic_string<char>, int>' to make the copy explicit or 'const std::pair<const std::__cxx11::basic_string<char>, int>&' to prevent copying
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Config.hpp:1611:57: warning: loop variable 'kvp' of type 'const std::pair<std::__cxx11::basic_string<char>, int>&' binds to a temporary constructed from type 'const std::pair<const std::__cxx11::basic_string<char>, int>' [-Wrange-loop-construct]
 1611 |             for (const std::pair<std::string, int32_t> &kvp : enum_keys_map)
      |                                                         ^~~
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Config.hpp:1611:57: note: use non-reference type 'const std::pair<std::__cxx11::basic_string<char>, int>' to make the copy explicit or 'const std::pair<const std::__cxx11::basic_string<char>, int>&' to prevent copying
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Config.hpp: In instantiation of 'static Slic3r::t_config_enum_names Slic3r::ConfigOptionEnum<T>::create_enum_names() [with T = Slic3r::SLADisplayOrientation; Slic3r::t_config_enum_names = std::vector<std::__cxx11::basic_string<char> >]':
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Config.hpp:1580:82:   required from 'std::string Slic3r::ConfigOptionEnum<T>::serialize() const [with T = Slic3r::SLADisplayOrientation; std::string = std::__cxx11::basic_string<char>]'
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Config.hpp:1577:17:   required from here
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Config.hpp:1607:57: warning: loop variable 'kvp' of type 'const std::pair<std::__cxx11::basic_string<char>, int>&' binds to a temporary constructed from type 'const std::pair<const std::__cxx11::basic_string<char>, int>' [-Wrange-loop-construct]
[... cut ...]
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Config.hpp:1611:57: note: use non-reference type 'const std::pair<std::__cxx11::basic_string<char>, int>' to make the copy explicit or 'const std::pair<const std::__cxx11::basic_string<char>, int>&' to prevent copying
make[2]: *** [src/slic3r/CMakeFiles/libslic3r_gui.dir/build.make:1266: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/Jobs/ArrangeJob.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1230: src/slic3r/CMakeFiles/libslic3r_gui.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
~~~

supermerill, 2021-09-27:
maybe it needs a `#include "libslic3r/ClipperUtils.hpp"` in ArrangeJob.cpp
i'll add it.

sl1pkn07, 2021-09-27:
hi

yes. that insert taht header worked, but now fails in the tests (  `-DSLIC3R_BUILD_TESTS=ON  -DBUILD_TESTING=ON`)

~~~make
[ 86%] Building CXX object tests/libslic3r/CMakeFiles/libslic3r_tests.dir/test_amf.cpp.o
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/tests/libslic3r/test_amf.cpp: In function 'void ____C_A_T_C_H____T_E_S_T____0()':
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/tests/libslic3r/test_amf.cpp:15:113: error: cannot convert 'Slic3r::Model*' to 'Slic3r::ConfigSubstitutionContext*'
   15 |             bool result_code = load_amf(get_model_path("test_amf/20mmbox_deflated.amf"s).c_str(), &_tmp_config, model, false);
      |                                                                                                                 ^~~~~
      |                                                                                                                 |
      |                                                                                                                 Slic3r::Model*
In file included from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/tests/libslic3r/test_amf.cpp:4:
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Format/AMF.hpp:10:95: note:   initializing argument 3 of 'bool Slic3r::load_amf(const char*, Slic3r::DynamicPrintConfig*, Slic3r::ConfigSubstitutionContext*, Slic3r::Model*, bool)'
   10 | extern bool load_amf(const char* path, DynamicPrintConfig* config, ConfigSubstitutionContext* config_substitutions, Model* model, bool check_version);
      |                                                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/tests/libslic3r/test_amf.cpp:24:128: error: cannot convert 'Slic3r::Model*' to 'Slic3r::ConfigSubstitutionContext*'
   24 |             bool result_code = load_amf(get_model_path("test_amf/20mmbox_deflated-in_directories.amf"s).c_str(), &_tmp_config, model, false);
      |                                                                                                                                ^~~~~
      |                                                                                                                                |
      |                                                                                                                                Slic3r::Model*
In file included from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/tests/libslic3r/test_amf.cpp:4:
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Format/AMF.hpp:10:95: note:   initializing argument 3 of 'bool Slic3r::load_amf(const char*, Slic3r::DynamicPrintConfig*, Slic3r::ConfigSubstitutionContext*, Slic3r::Model*, bool)'
   10 | extern bool load_amf(const char* path, DynamicPrintConfig* config, ConfigSubstitutionContext* config_substitutions, Model* model, bool check_version);
      |                                                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/tests/libslic3r/test_amf.cpp:33:124: error: cannot convert 'Slic3r::Model*' to 'Slic3r::ConfigSubstitutionContext*'
   33 |             bool result_code = load_amf(get_model_path("test_amf/20mmbox_deflated-mult_files.amf"s).c_str(), &_tmp_config, model, false);
      |                                                                                                                            ^~~~~
      |                                                                                                                            |
      |                                                                                                                            Slic3r::Model*
In file included from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/tests/libslic3r/test_amf.cpp:4:
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Format/AMF.hpp:10:95: note:   initializing argument 3 of 'bool Slic3r::load_amf(const char*, Slic3r::DynamicPrintConfig*, Slic3r::ConfigSubstitutionContext*, Slic3r::Model*, bool)'
   10 | extern bool load_amf(const char* path, DynamicPrintConfig* config, ConfigSubstitutionContext* config_substitutions, Model* model, bool check_version);
      |                                                                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/tests/libslic3r/test_amf.cpp:46:104: error: cannot convert 'Slic3r::Model*' to 'Slic3r::ConfigSubstitutionContext*'
   46 |             bool result_code = load_amf(get_model_path("test_amf/20mmbox.amf"s).c_str(), &_tmp_config, model, false);
      |                                                                                                        ^~~~~
      |                                                                                                        |
      |                                                                                                        Slic3r::Model*
In file included from /tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/tests/libslic3r/test_amf.cpp:4:
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Format/AMF.hpp:10:95: note:   initializing argument 3 of 'bool Slic3r::load_amf(const char*, Slic3r::DynamicPrintConfig*, Slic3r::ConfigSubstitutionContext*, Slic3r::Model*, bool)'
[... cut ...]
/tmp/makepkg/sl1-superslicer-git/src/SuperSlicer/src/libslic3r/Config.hpp:1611:57: note: use non-reference type 'const std::pair<std::__cxx11::basic_string<char>, int>' to make the copy explicit or 'const std::pair<const std::__cxx11::basic_string<char>, int>&' to prevent copying
make[2]: *** [tests/libslic3r/CMakeFiles/libslic3r_tests.dir/build.make:90: tests/libslic3r/CMakeFiles/libslic3r_tests.dir/test_amf.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:1471: tests/libslic3r/CMakeFiles/libslic3r_tests.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
~~~

supermerill, 2021-09-27:
then don't build the tests.

sl1pkn07, 2021-09-27:
is for package creation check

build ok not assume the program works if the check is not passed

greetings

Lukas1818, 2021-09-27:
The superslicer aur package does also hit a issue with the the test. But this was long time before. At this time the superslicer was still named slic3r++ and I was not maintainer of the aur package yet.
Disabling the test was not enough, they musst be removed from the cmake file.

The tests are still disable at the packges:
https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=superslicer#n31

The daily ci, wich does check the package, does not hit an error today:
https://gitlab.com/Lukas1818/aur-package-check/-/jobs/1623789318