supermerill/SuperSlicer #704: Brim Thickness ignoring first layer height and extruder height limits

bung69, 2020-11-22:
### Version
2.2.53

### Operating system type + version
Win10 1803

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behaviour
Brim Thickness is ignoring first layer height and extruder height limits, seems to be fixed to nozzle diameter.

nozzle 0.4, layer height max limits of 0.3, first layer height 50% = 
[image]

supermerill, 2020-11-22:
fixed in 2.2.53

bung69, 2020-11-24:
Perfect thank you.

closed 2020-11-29 by bung69