supermerill/SuperSlicer #705: Bridge infill fails to use straight lines, prioritizes curves

haleypearse, 2020-11-22:
### Version
_2.3.55

### Operating system type + version
Mac OS 10.15.6

### 3D printer brand / version + firmware version (if known)
Prusa m3ks,

### Behavior
Bridge infill fails to use straight lines, prioritizes curves. But a curved bridge is impossible. I noticed this in an earlier version of superslicer too, had to use prusaslicer. 

Superslicer:
[image]

I'm trying to print countersunk bolt holes btw. Another hole on the left is handled a bit better but still poorly. 
[image]

Prusaslicer:
[image]

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

haleypearse, 2020-11-22:
Oh man SuperSlicer is so awesome! I'm finding features I've wanted for ages, unconnected infill, supporting dense layer, even better than I imagined.

supermerill, 2020-11-23:
It's because of the "one peri on top". You ahve to deactivate it for this layer (or on the full print if it's not needed)

note that for replicating the prusa behavior on that, you need to set your `infill -> advanced -> Anchor solid infill by X mm -> DEfault & Bridged` to 3 (mm)

haleypearse, 2020-11-23:
Confirmed, that fixes it. Great to have so many configurable options and features. I'm like a kid in a candy shop. Thanks for your work on this. I hope I can contribute some day.

closed 2020-11-23 by haleypearse