supermerill/SuperSlicer #707: Enhancement: Add Custom G-Code tab to Print Profile

rngSplat, 2020-11-23:
### Version
2.2.53

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Heavily modified CR-10S, running Klipper

### Behavior
Requesting a custom G-Code option for print profiles, to facilitate calibration prints using Klipper's Tuning_Tower command (although there are probably other use-cases)

#### Project File (.3MF) where problem occurs
NA

supermerill, 2020-11-23:
https://github.com/supermerill/SuperSlicer/issues/282

> Requesting a custom G-Code option for print profiles

and it's trigger when/where?

rngSplat, 2020-11-24:
Most interested in start/end triggers. For example, I could make a print profile that included the gcode for pressure advance calibration, as described [here](https://www.klipper3d.org/Pressure_Advance.html).

I see that you marked it as duplicate. I searched for a similar one, but didn't see it. Happy to take my comments there, if you let me know what it's a duplicate of.

supermerill, 2020-11-25:
That's because there is already an issue for Pressure Advance calibration print, if it's the goal. Ultimately, if it's the only thing you need.

For inserting gcode to a print, you can right-click on the (+) 
[image]

rngSplat, 2020-11-27:
That method adds it to a single print, not a print profile. Having custom g-code for print profiles would make it easier and have fewer manual steps.

I did find issue #496 which this seems to be an exact duplicate of. I'll close this as a duplicate of that. (Sorry I didn't find that one earlier; I was looking for enhancement tag, not New Feature)

closed 2020-11-27 by rngSplat