supermerill/SuperSlicer #709: superslicer linux/ appimage now only functions with en_GB locales

shiftingtech, 2020-11-24:
### Version
2.3.55

### Operating system type + version
Linux (Debian)

### 3D printer brand / version + firmware version (if known)
Voron2.4

### Behavior
On first launch, the new appimage version of superslicer displays a dialog which states: 

Switching Superslicer to language en_GB failed.  You may need to reconfigure the missing locales, likely by running the "locale-gen" and "dpkg-reconfigure locales" commands

After hitting Okay on that dialog, SS exits immediately.

As a workaround, I used the second command  (dpkg-reconfigure locales) to generate a en_GB.utf8  locale, and after that, SS does run correctly, but I'm unsure why SS is unable to use other common english locales such as en_CA and en_US.  It's also worth noting that I've used a number of previous SS app images, and never encountered this problem.  It appears to be a recent development.

supermerill, 2020-11-25:
here's a `boost::nowide::nowide_filesystem();` that was added in august.
From when is you old version that works?

shiftingtech, 2020-11-25:
2.2.54 is what I had been using, which did not have the issue.

supermerill, 2020-11-26:
what's the value of your system variable $LC_ALL ?
If it's not set, try to `sudo systemctl set-environment LC_ALL=en_US.UTF-8` for using en_US

shiftingtech, 2020-11-27:
I did not have an $LC_ALL, however I tested a few variants  (en_US.UTF-8, en_CA.UTF-8) and it made no difference.  SS still fails if EN_GB.UTF-8 is unavailable.

EvilGremlin, 2020-11-27:
For me it's stuck in default system locale (russian) and cannot switch to english. No LC_ALL in system, setting it doesn't help.
I think you should make locale internal variable, completely independent from OS.

supermerill, 2020-11-28:
The software need to be in an 'en' locale for writing gcode stuff, to be able to use '.' as decimal separator.
Last time I tried to resolve a locale bug, it takes me one day to understand why it crashes and two other to find a hack...
For this one, I also need to reproduce it in the first place...

All that to say that it's a bugfix that will consume many many hours. I hope that it will be resolved by itself when the prusa  version will be out of beta.

Can you try to see if you have the same bug with the prusaslicer alpha-4?

shiftingtech, 2020-11-29:
I just tested the new prusaslicer beta1, since it's now out.
It suffers from the same bug.

supermerill, 2020-11-29:
Maybe you can report this bug on their github? I think they have more knowledge about it.