supermerill/SuperSlicer #712: Improvement :  Simple Wipe Tower

tirasec, 2020-11-24:
### Version
2.3.55

### Operating system type + version
Any

### 3D printer brand / version + firmware version (if known)
Dual head

### Behavior
When printing with Dual head and soluble support we have to clean the nozzle before each layer. 
It is possible to use the Ooze prevention feature but it creates a tall skirt all around the part. 
If the material is very expensive, like PEKK (~800€ per kg), and the object very big, it cost a lot of material.

For example this part  :
If I activate Ooze prevention (if not there is too much oozing on such kind of parts)
[image]
Material cost 183€ in PEKK/Soluble Support and printing time is 23h30
(for information, on large part, we can't use only one loop for the tall skirt, with one loop, the skirt fails most of times)

If I do not activate Ooze prevention :
[image]
Material cost 120.5€ in PEKK/Soluble Support and printing time is 18h30

I made simulation with a Wipe Tower manually designed, the material cost is 132€ (we save 50€ material compare to Ooze Prevention !) and printing time is 21h00
[image]

**Some ideas :**
- The wipe tower have to be up to the last extruder change (no need to go up to the top of the part, it save time and material)
- Some usefull parameters for the wipe tower : diameter, number of loops, brim size, plus maybe circular or rounded square shape
- Wipe tower could be as a 2 objects we can move, but always printed first after a tool change
[image]

**Some questions :**
- Is there a workaround to do that in SuperSlicer ?
- How can I be sure the WipeTower is printed before the object ?
- Could it be possible to generate the wipe tower with a FreeCad python script ? (unfortunately, I am not developer)

#### Project File (.3MF) where problem occurs
[WipeTower01.zip](https://github.com/supermerill/SuperSlicer/files/5589657/WipeTower01.zip)

tirasec, 2020-11-24:
After a few tests, it seems that if the wipe tower file is at the top of the parts list, it is printed first for each layer.
[image]

supermerill, 2020-11-25:
> Is there a workaround to do that in SuperSlicer ?
> How can I be sure the WipeTower is printed before the object ?

use the integrated wipe tower?

>  Could it be possible to generate the wipe tower with a FreeCad python script ? (unfortunately, I am not developer)

yes. if it's a cylinder, just write
```
cut()(
cylinder(myradius+0.45,myheight),
cylinder(myradius,myheight),
)
```

tirasec, 2020-11-25:
Thanks...  
Is there a way to use the integrated wipe tower with dual head printer ? Each time I activate the wipe tower, my second extruder switch to 0 !
Before activating integrated WipeTower (Ext1 is the Object, Ext2 is Support )
[image]

Message when I click on Enable Wipe Tower
[image]

After activating Support extruder switched to '0'
[image]

tirasec, 2020-11-26:
After testing, it is possible to workaround by this way :
- Import the STL of object
- Set print parameters and slice
- Look at the maximum height of support
- Import a STL file containing 2 concentric cylinders 
- Change the height of cylinders to the maximum height of support
- Move the cylinders on top of the list of files and set the print parameters (attach one extruder to each cylinder)
[image]
- Slice...
It's a bit long, but it works and has the advantage of being able to choose the size, position, and shape of the WipeTower...
For example, you can place the wipe tower inside the part...

corrado-c, 2022-04-30:
Hi! what I was looking for! I tried and it's not bad! a thousand thanks!
but then the thing evolved?