supermerill/SuperSlicer #713: PS alpha 4 crash issue heads up

neophyl, 2020-11-24:
Merill I know you have just done a new release against alpha 4 but I thought I'd give you a heads up in case you were unaware.  

There seems to be a LOT of issues being raised about alpha 4 just crashing on slicing.  Especially with gyroid infill.  I tried several of the linked 3mf and stl files and yes they all crash.  Unless I set the new PS anchored value to 0.  At that point they will slice fine.

The same files loaded into SuSie slice because that value is set to 0 by default.  If I set it to anything but 0 then SuSie also crashes.

supermerill, 2020-11-25:
I set this thing to 0 by default.
A good thing I kept the old codepath ;-p

neophyl, 2020-11-25:
I'll close it now I know you have seen it :)

closed 2020-11-25 by neophyl