supermerill/SuperSlicer #714: Fan kickstart time causing issues

ardyesp, 2020-11-24:
### Version
2.3.55.2

### Operating system type + version
Windows 7, Win64

### 3D printer brand / version + firmware version (if known)
Klipper on Custom CoreXY

### Behavior
If the `Printer -> General -> Cooling Fan` specifies a `Kickstart duration` which is not 0, the fan runs at 100% regardless of max/default fan speed in the Filament.

#### Project File (.3MF) where problem occurs
3DBenchy
Kickstart duration = 0.1, Fan running at 100% all the time
[image]

Kickstart duration = 0, Fan at 60% as configured, 100% over bridges and overhangs
[image]

supermerill, 2020-11-28:
can i have your .3mf?

ardyesp, 2020-11-29:
Sure, project file attached here, with XYZ calibration cube. The print should show 100% on overhangs, and 60% fan elsewhere.
Thanks.
[xyzCalibration_cube.zip.zip](https://github.com/supermerill/SuperSlicer/files/5612313/xyzCalibration_cube.zip.zip)

supermerill, 2020-11-29:
Seriously, i won't touch it anymore before my gcode rewrite.
You can disable it by putting 0 on it, and i'll change the default value to 0, so it will be deactivated.
I will be rewritten later.

ardyesp, 2020-11-29:
I understand. For now, I have set the fan kickstart parameter in the Klipper instead.

closed 2021-11-07 by supermerill