supermerill/SuperSlicer #718: ExportToPS Ironing problems (double layer prints?)

mocksoul, 2020-11-24:
### Version
ExportToPS Edition (2.3.55.2)

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
klipper / voron

### Behavior
Sometimes SuperSlicer tries to print layers twice oO.
In the included 3mf top prusa ironing made 2 times (with huge overextrusion).

[over_bridge_flow_test.zip](https://github.com/supermerill/SuperSlicer/files/5592886/over_bridge_flow_test.zip)

You can see this in this screen:

[image]

But if you change top pattern to "ironing" and turn off "ironing post process" (e.g. switch to superslicer ironing) - for some reason second layer is printed twice with same huge overextrusion:

[image]

Here two (both strange :)) results:

[image]

mocksoul, 2020-11-26:
Fixed in nightly? Should I try?

supermerill, 2020-11-26:
not yet. The commit has to show here, and then you can grab the nightly that will be built after that.

mocksoul, 2020-11-26:
Thanks! I see 👍

mocksoul, 2020-11-27:
Tried latest build. Original issue fixed, but still where is problems:

sometimes superslicer tries to put perimeters on top of prusa-style ironing layer. In the image below top ironing was done properly, but where is perimeter on top of ironing in the bottom of the image:

[image]

[image]

[front_rear_skirt_a_300_x2.zip](https://github.com/supermerill/SuperSlicer/files/5609478/front_rear_skirt_a_300_x2.zip)

closed 2021-06-04 by supermerill