supermerill/SuperSlicer #719: Concentric (filled) infill, gap fill misplaced

Zunzulla, 2020-11-25:
### Version
2.2.53
### Operating system type + version
Microsoft Windows 10 Pro 10.0.19041

### 3D printer brand / version + firmware version (if known)
Prusa MK3S 1.2mm nozzle

### Behavior
When slicing a circular object in vase mode with concentric (filled) infill, the gap fill appears to be misplaced

#### Project File (.3MF) where problem occurs

[image]
[image]

[Concentric(filled)IssueGapFill.zip](https://github.com/supermerill/SuperSlicer/files/5597739/Concentric.filled.IssueGapFill.zip)

Zunzulla, 2020-11-26:
Adding to this: using the concentric (filled) infill (same settings as in the project file posted before) on a square object, the center part is not filled. The gap fill only happens on one line, instead of all 4 
[image]

Zunzulla, 2020-11-26:
Adding the file 
[SquareConcentric(filled)IssueGapFill.zip](https://github.com/supermerill/SuperSlicer/files/5604213/SquareConcentric.filled.IssueGapFill.zip)

supermerill, 2020-11-26:
fixed in 2.3.55, have to port it back to the stable branch.

Zunzulla, 2020-12-02:
> fixed in 2.3.55, have to port it back to the stable branch.

Sorry, didn't understand ^^ I tried slicing with the latest stable branch and 2.3.55, the gap fill still gets misplaced

supermerill, 2020-12-03:
Oh, it's a vase mode bug

closed 2024-02-11 by supermerill