supermerill/SuperSlicer #721: Cooling Object Fan management with Dual Head printer

tirasec, 2020-11-25:
### Version
2.3.55.2

### Operating system type + version
W10

### 3D printer brand / version + firmware version (if known)
Dual Head printer / Duet3D + Reprap Firmware 3.1.1

### Behavior
When printing object with Dual head. We have to setup one fan for cooling object and one other fan for cooling support. Parameters of the fans are different depending on the type of material/support used.
On RepRapFirmware,  we have to know the fan we handle, the GCode command is **M106 P0 Sxxx** for the fan cooling the object (T0 tool) and **M106 P1 Sxxx** for the fan cooling the support material (T1 tool).
Is there a way or workaround to set up the T0 fan and T1 fan according to the cooling parameters in filament settings?

#### Project File (.3MF) where problem occurs

supermerill, 2020-11-26:
This isn't done automatically?

tirasec, 2020-11-26:
Looks no, on the GCode, I have only M106 Sxxx values... No "P" value (of course, I am on RepRapFirmware GCode flavor ) 
I searched, but didn't find a place to configure the GCode for the fans.
Would it be possible with a post-processing script or something like that?

closed 2021-06-04 by supermerill