supermerill/SuperSlicer #723: XY compensation

Slatkey, 2020-11-26:
### Version
2.3.55.2

### Operating system type + version
Windows 10 64

### 3D printer brand / version + firmware version (if known)
Corexy / Klipper

### Behavior
Hello. Artifacts appear on the model when XY compensation is turned on.

#### Project File (.3MF) where problem occurs

[z_drive_retainer_b_x2.zip](https://github.com/supermerill/SuperSlicer/files/5602503/z_drive_retainer_b_x2.zip)

Without compensation:
[image]
[image]

With compensation included
[image]
[image]

supermerill, 2020-11-26:
Where is the artifact? can't spot it

Slatkey, 2020-11-26:
[image]

closed 2021-06-04 by supermerill

TheDoc13, 2022-12-23:
Sorry to revive a presumably dead issue, but I got the same problem in V2.4.58.5
[image]

Layer 71 (0.2mm layer height) is using the compensation, everything below not)

Is the bug back?