supermerill/SuperSlicer #726: Crash to desktop on slicing a file

LordMelkor, 2020-11-26:
### Version
SuperSlicer_2.3.55.2_win64_20201124

&

SuperSlicer_2.3.55.1_win64_20201118

### Operating system type + version
Win 10 Pro 64bit

### 3D printer brand / version + firmware version (if known)
MK3S

### Behavior
* The attached 3MF file was created in PrusaSlicer alpha4. PrusaSlicer alpha4 crashes when trying to slice this project. SuperSlicer 2.3.55.2 also crashes when trying to slice this project.

However, **SuperSlicer 2.3.55.1 can successfully slice this project.**

PrusaSlicer Alpha4 was originally used because it supports auto arranging parts on the build plate _with rotations_.

#### Project File (.3MF) where problem occurs
[Filament_Spool_Disk.zip](https://github.com/supermerill/SuperSlicer/files/5605259/Filament_Spool_Disk.zip)

[image]

supermerill, 2020-11-26:
using 0 in "Length of Infill Anchor" makes the slicer use the good old algorithm for linking infill lines.
If you put an other value, it will use the new totally buggy one. #713

darkspawn91, 2020-12-08:
I found a solution, if you change “Lenght of the infill anchor” in my case from “0” to “2mm” you have to change also Sparse infill, and set it to the drop-down menu. I chose 23%, no crash came out. BTW I don’t know why.

supermerill, 2020-12-08:
> I don’t know why.

luck

darkspawn91, 2020-12-08:
Sorry Merrill, I’m trying to solve the over extrusion on perimeters (buldge?) without success.
Can you explain me the difference between “Anchor solid infill by X mm” and “infill overlap”, please? 
Thanks again

supermerill, 2020-12-08:
* `Infill/perimeter overlap` will stretch the infills (sparse & solid) to push it into the most inner perimeter.
* `Anchor solid infill by X mm` 
  * `Default`: Is the amount of extra mm the solid area is allowed to expand into the sparse infill, to be able to support the perimeters & to be well supported by the previous layer. 
  * `bridge` it's to set the distance the bridge area can enter into the not-bridge area.

tooltip are not enough? where can i improve them?
try with high/low value, it's easy to spot the differences.