supermerill/SuperSlicer #729: Crash while slicing

lovesegfault, 2020-11-28:
### Version
2.3.55.2

### Operating system type + version
NixOS Unstable

### Behavior
Trying to slice [this part](https://github.com/VoronDesign/Voron-0/blob/master/VORON-0/STLs/Skirt_x3.STL) gives me a crash. Here's the backtrace:
```
#0  0x000000000094f1b7 in Slic3r::FakePerimeterConnect::connect_infill(std::vector<Slic3r::Polyline, std::allocator<Slic3r::Polyline> >&&, std::vector<Slic3r::Polygon const*, std::allocator<Slic3r::Polygon const*> > const&, Slic3r::BoundingBox const&, std::vector<Slic3r::Polyline, std::allocator<Slic3r::Polyline> >&, double, Slic3r::FillParams const&) ()
No symbol table info available.
#1  0x000000000095096c in Slic3r::FakePerimeterConnect::connect_infill(std::vector<Slic3r::Polyline, std::allocator<Slic3r::Polyline> >&&, Slic3r::ExPolygon const&, std::vector<Slic3r::Polyline, std::allocator<Slic3r::Polyline> >&, double, Slic3r::FillParams const&) ()
No symbol table info available.
#2  0x000000000095f324 in Slic3r::FillGyroid::_fill_surface_single(Slic3r::FillParams const&, unsigned int, std::pair<float, Slic3r::Point> const&, Slic3r::ExPolygon, std::vector<Slic3r::Polyline, std::allocator<Slic3r::Polyline> >&) const ()
No symbol table info available.
#3  0x000000000094645c in Slic3r::Fill::fill_surface(Slic3r::Surface const*, Slic3r::FillParams const&) const ()
No symbol table info available.
#4  0x00000000009466db in Slic3r::Fill::fill_surface_extrusion(Slic3r::Surface const*, Slic3r::FillParams const&, std::vector<Slic3r::ExtrusionEntity*, std::allocator<Slic3r::ExtrusionEntity*> >&) const ()
No symbol table info available.
#5  0x000000000091b12e in Slic3r::Layer::make_fills(Slic3r::FillAdaptive::Octree*, Slic3r::FillAdaptive::Octree*) ()
No symbol table info available.
#6  0x00000000007995e4 in tbb::interface9::internal::start_for<tbb::blocked_range<unsigned long>, Slic3r::PrintObject::infill()::{lambda(tbb::blocked_range<unsigned long> const&)#1}, tbb::auto_partitioner const>::execute() ()
No symbol table info available.
#7  0x00007f30ef7f72e5 in tbb::internal::custom_scheduler<tbb::internal::IntelSchedulerTraits>::process_bypass_loop(tbb::internal::context_guard_helper<false>&, tbb::task*, long) ()
   from /nix/store/l5g87w4ahbbmvvq9xjwlqg5bxf3x16kh-tbb-2019_U9/lib/libtbb.so.2
No symbol table info available.
#8  0x00007f30ef7f75ab in tbb::internal::custom_scheduler<tbb::internal::IntelSchedulerTraits>::local_wait_for_all(tbb::task&, tbb::task*) () from /nix/store/l5g87w4ahbbmvvq9xjwlqg5bxf3x16kh-tbb-2019_U9/lib/libtbb.so.2
No symbol table info available.
#9  0x00007f30ef7f0d97 in tbb::internal::arena::process(tbb::internal::generic_scheduler&) () from /nix/store/l5g87w4ahbbmvvq9xjwlqg5bxf3x16kh-tbb-2019_U9/lib/libtbb.so.2
No symbol table info available.
#10 0x00007f30ef7ef840 in tbb::internal::market::process(rml::job&) () from /nix/store/l5g87w4ahbbmvvq9xjwlqg5bxf3x16kh-tbb-2019_U9/lib/libtbb.so.2
No symbol table info available.
#11 0x00007f30ef7ec0bc in tbb::internal::rml::private_worker::run() () from /nix/store/l5g87w4ahbbmvvq9xjwlqg5bxf3x16kh-tbb-2019_U9/lib/libtbb.so.2
No symbol table info available.
#12 0x00007f30ef7ec2d9 in tbb::internal::rml::private_worker::thread_routine(void*) () from /nix/store/l5g87w4ahbbmvvq9xjwlqg5bxf3x16kh-tbb-2019_U9/lib/libtbb.so.2
No symbol table info available.
#13 0x00007f30ee299e9e in start_thread (arg=0x7f3087fff640) at pthread_create.c:463
        ret = <optimized out>
        pd = 0x7f3087fff640
        unwind_buf = {cancel_jmp_buf = {{jmp_buf = {139846416856640, 1647462900904908214, 139846391670558, 139846391670559, 0, 52245520, -1675909466604546634, -1676136668463475274}, mask_was_saved = 0}}, priv = {pad = {0x0, 0x0, 0x0,
              0x0}, data = {prev = 0x0, cleanup = 0x0, canceltype = 0}}}
        not_first_call = 0
#14 0x00007f30ee1c866f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
```

neophyl, 2020-11-28:
Probably same as #726. It’s the same issue the Prusa slicer alpha 4 keeps crashing, seeing as that version is the one that 2.3.55.2 is built against its inherited the same issue.

supermerill, 2020-11-28:
[image]
I need your project file

lovesegfault, 2020-11-29:
@supermerill Here you go, hopefully it helps with reproducing: [skirt_project.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5611554/skirt_project.3mf.zip)

supermerill, 2020-11-29:
Seems fixed on the latests nightly. Can you download i and test?

lovesegfault, 2020-11-30:
Yup, seems to have been fixed!

Thanks!!

closed 2021-06-04 by supermerill