supermerill/SuperSlicer #731: [FEATURE REQUEST] Filament change per layer based on filament profile

wildjokerss, 2020-11-29:
### Version
2.3.55.2

### 3D printer brand / version + firmware version (if known)
Marlin

### Behavior
Problem: Whenever I do filament change at a certain layer we can send either custom g-code or simply add the color change with the M600 feature. Unfortunately this gets a bit more painful if we're doing color change to different filaments, for example, I could be printing PETG at 250, them want some PLA at 220 or 200, then some TPU at whatever temperature it is.
Even if the changes are all within the same type of filament, e.g TPU, if they are different brands they can require different temps.
This is obviously not too much trouble if you want to do one or two color changes, but adding the g-code for many becomes annoying.

My idea would be to have something similar to multi color printing, whenever we add a "M600" we have a drop down on the list to pick which filament we are printing. This would allow to have specific temperature and fan speed from that filament on the color change. Obviously the bed temperature should stay the same for the first filament. 

I'm unsure if I was clear but I hope I was, let me know if I can explain anything better :)

supermerill, 2020-11-29:
* add extruders to your printer
* change extruders as you would do with a real multi-extruder printer
* put M600 in the toolchange gcode

wildjokerss, 2020-11-29:
Hi @supermerill, that could indeed to the trick it seems. 
I was investigating this and I'm left with two questions:
1 - I need a temperature change to take place before the M600, so that the new filament starts on it's ideal temp, is there a list of variables so I know how to get it working? Ill need like a "next extruder filament temp" so if I change from e.g PETG to PLA, I won't have PLA kicking in at ~250ºC
2 - Was looking at the G-code preview and it is setting the temperature of the next filament 2 layers earlier, I believe this could cause some issues, specially going from hotter to colder. Screenshot and 3MF attached
[image]
[too early temperature change.zip](https://github.com/supermerill/SuperSlicer/files/5612316/too.early.temperature.change.zip)

Also, I'm not sure I want a wipe tower, what makes you suggest to turn it on? I'm usually fine with the M600 purge

supermerill, 2020-11-29:
I looked at the gcode and it's all good.
It's a bug in the new gcode viewer.

wildjokerss, 2020-11-29:
In that case that sounds fine. My bad, should have checked the G-code.
What about the temperature?

_; custom gcode: end_filament_gcode
; Filament-specific end gcode 
;END gcode for filament
; custom gcode end: end_filament_gcode
**; custom gcode: toolchange_gcode
M600**
; custom gcode end: toolchange_gcode
M107
T1
**M104 S285 ; set temperature**
; custom gcode: start_filament_gcode
; Filament gcode
; custom gcode end: start_filament_gcode
; printing object Body1.stl id:0 copy 0_

Looking at the G-code above it is still "unfortunate" that the temperature change takes place only after the M600. 
If the temperature change were to take place immediately before the M600 command, the hotend would get closer to the desired temperature when it starts to print with the next filament. 
Is there any way from the slicer this could be turned around or needs to be in the code?

neophyl, 2020-11-29:
https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-placeholders-(a-copy-of-the-mauk.cc-page) is Prusa Slicers list of placeholders (not sure if its a complete list though).

Theres an exaple
"next_extruder
M104 S[first_layer_temperature_[next_extruder]]
note: this only works in tool change gcode"  

maybe look at that as it allow next/previous extruder with the temperature variables so if you place the appropriate line before your M600 in the tool change ?

wildjokerss, 2020-11-29:
@neophyl that seems quite awesome! Now just need to check if it works but if it does, it is wonderful, thank you!

wildjokerss, 2020-11-29:
; custom gcode: toolchange_gcode
M104 S220
M600
; custom gcode end: toolchange_gcode

thank you once again @neophyl , works wonders!

supermerill, 2020-11-29:
imo, it's better to use the temperature instead of the first layer one:

`M104 S{temperature[next_extruder]}`

wildjokerss, 2020-11-30:
hi @supermerill 
Been testing this and came across a unfortunate outcome.
The tool change G-code runs on the beginning of the print as well, so before starting to print it sets the temperature to the next extruder and runs the M600 :( this becomes a bit less practical 

I've attached a project and g-code generated by it if you wanna take a look.
[sad.zip](https://github.com/supermerill/SuperSlicer/files/5614765/sad.zip)

I've tried {if layer_z > 2} to skip any code on the early stages of the print but this does not seem to slice, producing an error. 

Am I missing anything or is there anything else I can do to avoid this?

wildjokerss, 2020-11-30:
okay, this went super bad
@supermerill the toolchange does not seem to be an option.
In theory my G-code was intended to start blue -> red strip -> Blue again -> red strip -> blue again
I understand a need of **4** M600 commands. 
The g-code on the tool change generated **41** of them 😬 😂 

Below the 3MF alongside the g-code generated by SS 2.3.55.3
[m600 party.zip](https://github.com/supermerill/SuperSlicer/files/5615135/m600.party.zip)

wildjokerss, 2020-11-30:
It's me again :D
After dragging my head around with this I found a fix for the huge amount of M600 on the code.
[image]
[image]
[image]

This is looking way better. **5** M600 found in the g-code, where 2 are just comments so left with 3
Current problem is how to stop the M600 from being added on the very beginning of the g-code file, as manual removal is a bit tedious.

Suggestions?

supermerill, 2020-11-30:
```
{if layer_num> 0}M104 S{temperature[next_extruder]}{endif}
{if layer_num> 0}M600{endif}
```

Imo, with only m600, there is too much ooze, and there is some defects that can happen. It's more useful when you're sapping for a top surface only, as for your cas eit will be hidden in an internal perimeter .
also uncheck the print->multipleextruders->wipetower -> prime all printing extruders

wildjokerss, 2020-11-30:
Did it! Works wonders! wew 😅

wildjokerss, 2020-12-06:
I guess I can close this, past couple of days been messing around and seems to be great this way 👍

closed 2020-12-06 by wildjokerss