supermerill/SuperSlicer #732: Infill showing on part perimeters

neophyl, 2020-11-29:
### Version
2.2.55.1

### Operating system type + version
Win10 64

### 3D printer brand / version + firmware version (if known)
MK3 with a geared extruder

### Behavior
I was printing some parts with SuSie when I noticed that the cubic infill pattern was visible on the surface.  It was very bad.
[image]
(this is from a cr10 with a 0.6 nozzle.  It could probably do with some tuning though).

Given how apparent it was I decided to do some comparison testing using my Prusa as that generally gives me nice prints.

So I added a simple test object and using profiles as close as I could get between Susie and PS 2.3beta (latest one I had) I printed the same from both silcers.  Same filament, one after the other.  

This was the result-
[image]

Susie on the left and PS on the right.  On the one from PS you can very faintly see the pattern, in person this looks almost like a minuscule depression in the surface but its only visible when the light hits it at the right angle.  The Susie part the infill pattern is actually raised above the surface and sticks out.  Its quite noticeable.

I think with the 0.6 nozzle it is more pronounced.

I have attached the projects for the test piece.  Hopefully its just a setting somewhere that needs altering.

#### Project File (.3MF) where problem occurs
[SS_artefact test 1.zip](https://github.com/supermerill/SuperSlicer/files/5612251/SS_artefact.test.1.zip)

supermerill, 2020-11-29:
Seriously, i won't touch it anymore before my gcode rewrite.
You can disable it by putting 0 on it, and i'll change the default value to 0, so it will be deactivated.
I will be rewritten later.

neophyl, 2020-11-29:
This issue is in 2.2.55.1 - so doesnt have the value in for setting to zero like 2.2.55.2.  It only has the 4 options for connected, connected to hole, connected to outer and not connected.  With the alpha issues in 2.2.55.2 I've removed it from my computer.

I'm just trying the same item and settings with 2.2.53.4 stable to see what that gives.

supermerill, 2020-11-29:
Sorry, i have written a comment in the wrong thread.
I'll look at it, but i have to print these print first