supermerill/SuperSlicer #733: Possible bug with "only one perimeter in top layer" turned on

wildjokerss, 2020-11-29:
### Version
2.3.55.2

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Marlin on custom Hypercube

### Behavior
* _Describe the problem_ only one perimeter in top layer seems to mess up the top layer itself
* _Steps needed to reproduce the problem_ enabling "only one perimeter in top layer"
* _Expected Results_ for everything to be smooth
* _Actual Results_ weird "holes" on top surface

  * _Screenshots from __*SuperSlicer*__ preview are preferred_
[image]
"only one perimeter on top surface" enabled

[image]
"only one perimeter on top surface" disabled

#### Project File (.3MF) where problem occurs
[only one perimeter on top surface - weird behaviour.zip](https://github.com/supermerill/SuperSlicer/files/5612274/only.one.perimeter.on.top.surface.-.weird.behaviour.zip)

supermerill, 2020-11-29:
put the min surface to 0%

the problem here is that the letters are grown, merged, and then shrinks, but as they merged, the shrunk didn't succeed to make it disappear as it should. Not that easy to do...

wildjokerss, 2020-11-29:
tried that, works at 0% as well as 100%. seems like a fix, thanks @supermerill !

supermerill, 2020-11-29:
found a way

closed 2021-06-04 by supermerill