supermerill/SuperSlicer #734: Feature Request: Extra length on restart - Option to define restart length as a function of travel distance

JasonGuthrie, 2020-11-30:
Hi, this suggestion was born from a Facebook forum discussion regarding PETG ooze (and subsequent under-extrusion) over long travel lengths and how to best cope with it. 

The other party in the conversation discussed how he had attempted to manage the issue over longer runs by compensating for ooze induced under-extrusion with extra restart distance, but quickly realised that this just caused over extrusion in parts with shorter travel moves. 

His ultimate solution was to rearrange the models to minimise travel length, however I was thinking a more elegant (and effective solution) would be to define the extra restart length as a function of the travel move e.g. 1mm + 0.1mm per 20mm travel length (or such like). 

I suspect the ooze is non-linear in nature, so trying to overcome it using a single constant probably isn't the most ideal solution (my guess is it's some asymptotic function like the exponential, but that's a wild stab in the dark!).

Let me know what you think,
Cheers,
Jason G

### Version
_Version of SupserSlicer used goes here, Use `About->About SuperSlicer`_
Any
### Operating system type + version
_What OS are you using (and version)_
W10 Pro - Up to date
### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Prusa - V6 (stock)
### Behavior
* _Describe the problem_

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_

supermerill, 2020-11-30:
> Let me know what you think,

I don't know. Someone need to make tests.
like two pieces with on1 perimeter, no infill, and see how the distance impact the perimeter.

then, find the best 'Extra length on restart' for each travel distance.
Then, you can extrapolate the function, 
* maybe ax²+bx+c
* or ax until x=b then ab
* or something else

JasonGuthrie, 2020-11-30:
> I don't know. Someone need to make tests.
> like two pieces with on1 perimeter, no infill, and see how the distance impact the perimeter.
>
> then, find the best 'Extra length on restart' for each travel [...]

That seems like a logical way of experimenting. 

I'll be keen to undertake the test but it will have to wait until after New Year.

supermerill, 2020-11-30:
I'll do the feature when the result will comes, no worry.

JasonGuthrie, 2020-11-30:
> I'll do the feature when the result will come, no worry.

Super!

robthide37, 2020-12-12:
If you need help testing i have dragon high flow and a standard one and a mosquito standard and e3d all on diff printers but the dragons are on same model printer so i can print whatever test you devise to try get more data

JasonGuthrie, 2020-12-13:
> If you need help testing i have dragon high flow and a standard one and a mosquito standard and e3d all on diff printers but the dragons are on same model printer so i can print whatever test you devise to try get more data

Cool, I get back to you just after the New Year (i've set aside time on my calendar for this experiment for the first two weeks of January) and we can discuss the best plan of attack. 

Speak to you later and happy Christmas/New Year!

Roskott, 2022-12-25:
If anyone would like to collaborate on testing this, I would really like to do so. 

This feature, while fixing issues I have with the external perimeter on PETG prints, also causes another issue with smaller extrusions such as gap fill where the extra restart length causes blobs to build up and potentially knock the part off or push out of register, which is predictably playing havoc, especially when attempting to print gears or other fine detailed mechanisms.