supermerill/SuperSlicer #736: Hotend on at the start of printing even without being configured for it

Rodrigormaraujo, 2020-11-30:
### Version
Version: 2.3.55.2
Build: SuperSlicer_2.3.55.2

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 8,458MB

### 3D printer brand / version + firmware version (if known)
Duet 2 wifi 1.04 - RepRapFirmware for Duet 2 WiFi/Ethernet 2.05 (2019-12-13b1)

### Behavior
**This is my start gcode script:**
M190 S{first_layer_bed_temperature[0]}  ; wait for bed temperature
G28
G1 X0 Y0 Z2 F1000
M400
M109 S[first_layer_temperature]             ; wait for nozzle temperature
G1 X0 Y0 Z0 F1000.0 ; move out of print volume
G1 Y10 E9 F500.0 ; start purge
G1 Y50 E9 F500.0 ; finish purge line

I don't know what's going on, but the hotend has been warming up since the beginning, along with the table. I don't know if there is any option that I enabled wrong. The hotend should only heat up after the table is hot and after the G28.

#### Project File (.3MF) where problem occurs
[AnevoMGN9H2.zip](https://github.com/supermerill/SuperSlicer/files/5616211/AnevoMGN9H2.zip)

neophyl, 2020-11-30:
I sliced your project and looked at the output gcode and there arent any extra heating commands in there that I can find.

In fact as your start gcode doesnt contain the lines for heat up bed to first layer temp and wait it only heats up to the final printing temp so the lower first layer bed temp you have defined in the filament is not being used.

One thing to investigate is the Duet Reprap implementation webpage states that the M command ar deprecated and that it should be using G10 followed by M116 to get the same effect.  It may be that your firmware is not behaving properly and waiting for those commands.  Something to take up with the firmware developers.

Rodrigormaraujo, 2020-11-30:
> I sliced your project and looked at the output gcode and there arent any extra heating commands in there that I can find.
> 
> In fact as your start gcode doesnt contain the lines for heat up bed to first layer temp and wait it only heats up to the final printing temp so the lower first layer bed temp you have defined in the filament is not being used.
> 
> One thing to investigate is the Duet Reprap implementation webpage states that the M command aer deprecated and that it should be using G10 followed by M116 to get the same effect.
> https://duet3d.dozuki.com/Wiki/Gcode#Section_M109_Set_Extruder_Temperature_and_Wait
> It may be that your firmware is not behaving properly and waiting for those commands. Something to take up with the firmware developers.

The only test I did was slicing using Prusa Slicer 2.2 and Simplify3d 4.0, and I didn't have this problem. The scripts are exactly the same, changing only the names of the hotend and bed variables.

neophyl, 2020-11-30:
Super Slicer iirc has been updated to be in line with the current reprap firmware standard so theres a G10 that's been inserted at the top of the output gcode file.  I think that is what is actually heating up your extruder.  Basically if you dont add in suitable start gcode for your flavour then it will auto insert it.

As reprap no longer uses the M109 command that you inserted in your start gcode SuSie is adding in a G10 for you.
I think if you comment out the M109 and replace it with - 
G10 [first_layer_temperature]
M116 ;although Im not 100% sure on this as Im just going by the duet/reprap page

Then Susie wont add one in for you at the top and it *should* work like you want.

neophyl, 2020-11-30:
The only work around I can come up with is to add another at the top that immediate sets the extruder temps to zero before starting to heat the bed and then heating the extruder like so -

G10 P0 S0 R0 ;  to counter the one inserted at the start of gcode generation for some reason.

M190 S[first_layer_bed_temperature]  ; wait for bed temperature
G28
G1 X0 Y0 Z2 F1000
M400
;M109 S[first_layer_temperature]             ; wait for nozzle temperature - deprecated on Duet RepRap

G10 P0 S[first_layer_temperature] R[first_layer_temperature]
M116 ;although I'm not 100% sure on this as Im just going by the duet/reprap page

G1 X0 Y0 Z0 F1000.0 ; move out of print volume
G1 Y10 E9 F500.0 ; start purge
G1 Y50 E9 F500.0 ; finish purge line

supermerill, 2020-11-30:
Maybe I have to remove the SXX part on these lines.
If I only print `G10 P0 R230 ; sets the standby temperature` it's correct?

Rodrigormaraujo, 2020-11-30:
G10 P0 R230 set the stand by temperature, not the active temperature.
[image]

Rodrigormaraujo, 2020-11-30:
> The only work around I can come up with is to add another at the top that immediate sets the extruder temps to zero before starting to heat the bed and then heating the extruder like so -
>
> G10 P0 S0 [...]

Your suggestion does, until the problem is resolved. Thank you for your attention to the matter!

supermerill, 2020-11-30:
> G10 P0 R230 set the stand by temperature, not the active temperature.

I'll understand that as "yes that's the fix i requested" and push it in the next release

Rodrigormaraujo, 2020-11-30:
> > G10 P0 R230 set the stand by temperature, not the active temperature.
> 
> I understand that as "yes that's the fix i requested" and will push it in the next release

That's right, thank you very much!

TypQxQ, 2021-01-16:
> Maybe I have to remove the SXX part on these lines.
> If I only print `G10 P0 R230 ; sets the standby temperature` it's correct?

It is better not to include any if there is one "G10 P" in the start G-Code. Also for a toolchanger the code below works better as you would want to set the temperature on the extruder actually used and not just the first one.
`G10 P[initial_extruder] R{filament_toolchange_temp[initial_extruder]} S{first_layer_temperature[initial_extruder]+extruder_temperature_offset[initial_extruder]}`