supermerill/SuperSlicer #737: 2.2.54: Removing Unsupported Perimeters from Bridge Area

CCS86, 2020-11-30:
2.2.54 / Win10 64

I know that running such a large number of perimeters isn't ideal. but, in this case it gets me a much better result, compared to less perimeters and concentric infill.

The only trouble it gives me is in this bridge area. I would think that the "remove perimeters" setting would remove the area with the red arrows in favor of bridge paths. But, I have tried all settings there, and no change is made.

[image]

Am I misunderstanding this feature, or could this be a bug?

[OP10-Fix.3mf.txt](https://github.com/supermerill/SuperSlicer/files/5617805/OP10-Fix.3mf.txt)

supermerill, 2020-12-03:
That feature was made for printing bolt-hole, where you have completely unsupported perimeters.
I didn't checked (yet) why the algorithm do that in your case.
Also, you can use less perimeters, at least on this layer.

CCS86, 2020-12-03:
Awesome, thanks for that anchor tip. This looks much better:

[image]

Is there a way to change the perimeter count for a single layer?

neophyl, 2020-12-03:
Normally you would use a modifier and set the perimeters to whatever you want for that layer. Either. Height range modifier or a normal one.