supermerill/SuperSlicer #739: Verbose is not appearing with G-Code for octolapse

bangor-flyer, 2020-12-01:
### Version
SuperSlicer
Version:   2.3.55.2
Build:     SuperSlicer_2.3.55.2

### Operating system type + version
_Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 17,128MB
OpenGL installation
GL version:   4.6.0 NVIDIA 456.38
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 770/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
When compiling with verbose ticked for octolapse it does not seem to recognize that its there.
after checking the g-code i can see the descriptive text.
Re-attempted with previous version of super slicer and no issues...

      
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*SuperSlicer*__ preview are preferred_
[image]
[image]

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_

supermerill, 2020-12-02:
can i have your project file & your gcode fie?

bangor-flyer, 2020-12-02:
Also the version of octolapse and octoprint 
[image]
[image]

FormerLurker, 2021-01-04:
I looked through the verbose output and there have been several changes, and what appears to be several errors.  See these verbose comments for 'skirt' gcode:

```
G1 X115.038 Y139.338 F7800.000 ; move to first skirtSkirt point
G1 X114.992 Y139.377 E18.74925 ; skirtSkirt
```

The other change that has broken Octolapse is a change of case.  I'm going to make the comparison case insensitive.  However, you may want to re-examine your verbose comment cases in general because I am also seeing strings like this:

```
; move to first Perimeter point
```

I REALLY like it that you added this:

```
G1 X178.665 Y151.310 E0.35072 ; External perimeter
```

because that will improve quality with Octolapse once I make the tests case insensitive.  While you are at it, it might not hurt to change these two:

```
G1 X167.454 Y148.472 ; move to first Perimeter point
G1 X167.804 Y148.225 E0.01233 ; Perimeter
```

to 
```
G1 X167.454 Y148.472 ; move to first internal perimeter point
G1 X167.804 Y148.225 E0.01233 ; internal perimeter
```

in order to distinguish internal and external perimeters.  In Slic3r and PrusaSlicer only Perimeter is used, so these can't be classified as definitely internal or external perimeters.  Adding the 'Internal' will allow this to be 100% compatible with the other existing Slic3r variants.

Finally, if you have a list of all possible verbose comments, I'd love to see them.  I've been looking through some examples, and have found many new feature tags that seem quite useful.  Thanks!

Thanks!

orecus, 2021-02-28:
Having the same issue, would be great if this could be fixed!

The-EG, 2021-03-10:
@FormerLurker 
> Finally, if you have a list of all possible verbose comments, I'd love to see them

I started poking around with this and found:
https://github.com/supermerill/SuperSlicer/blob/master/src/libslic3r/ExtrusionEntity.cpp#L267-L293

```c++
std::string ExtrusionEntity::role_to_string(ExtrusionRole role)
{
    switch (role) {
        case erNone                         : return L("Unknown");
        case erPerimeter                    : return L("Perimeter");
        case erExternalPerimeter            : return L("External perimeter");
        case erOverhangPerimeter            : return L("Overhang perimeter");
        case erInternalInfill               : return L("Internal infill");
        case erSolidInfill                  : return L("Solid infill");
        case erTopSolidInfill               : return L("Top solid infill");
        case erIroning                      : return L("Ironing");
        case erBridgeInfill                 : return L("Bridge infill");
        case erInternalBridgeInfill         : return L("Internal bridge infill");
        case erThinWall                     : return L("Thin wall");
        case erGapFill                      : return L("Gap fill");
        case erSkirt                        : return L("Skirt");
        case erSupportMaterial              : return L("Support material");
        case erSupportMaterialInterface     : return L("Support material interface");
        case erWipeTower                    : return L("Wipe tower");
        case erMilling                      : return L("Mill");
        case erCustom                       : return L("Custom");
        case erMixed                        : return L("Mixed");
        default                             : assert(false);
    }

    return "";
}
```

Not an exhaustive list, but that's the overall types. 

Also note, while making them lower case does allow the release version of OctoLapse to correctly parse a test file again (I couldn't reproduce the duplicating), it messes up the by "Feature Type" view in the gcode preview (and who knows what else):
[image]

So I would bet those strings are tied to multiple things in a case sensitive way.

supermerill, 2021-06-03:
sorry for the delay, I took my time going trough some months of issues that were pilling up.

@FormerLurker 
> G1 X115.038 Y139.338 F7800.000 ; move to first skirtSkirt point
> G1 X114.992 Y139.377 E18.74925 ; skirtSkirt

fixed, it's now `move to first Skirt point` and `; Skirt`

also fixed the `; move to first Brim point`

> in order to distinguish internal and external perimeters. 

Ok I change Perimeter by Internal perimeter.

note that since the 2.3 version (prusaslicer and superslicer), we have 
```
;TYPE:Solid infill
;WIDTH:0.45
```
at each extrusion change, so you don't need verbose output anymore

FormerLurker, 2021-06-03:
> ;TYPE:Solid infill
> ;WIDTH:0.45

That is AWESOME!!!!!  I LOVE this, AND I wasn't aware of this change in prusaslicer/superslicer....  Adding to milestones because that's huge!

closed 2021-06-13 by supermerill

vpicone, 2021-11-03:
@supermerill sorry, just to be clear, verbose g-code is no longer required for OctoLapse to know the currently printing feature? If so, I can open a PR to update [their docs](https://github.com/FormerLurker/Octolapse/wiki/V0.4---Slic3r-Slic3r-PE-and-PrusaSlicer#enabling-verbose-gcode). Would be super nice to not have "verbose" clones of all the default profiles.

supermerill, 2021-11-04:
It's not needed anymore, but I don't know if octolapse was updated to take the new comments into account.

vpicone, 2021-11-04:
I'll check it out, you're awesome. Thanks for all your work!