supermerill/SuperSlicer #741: export G-code for SLA printer

ogiwolf, 2020-12-01:
### Version
SuperSlicer_2.2.53.4

I am new to 3D printing
Just wondering can SLA printer export a G-code file?
When I am using the SuperSlicer, it can only export sl1 files.

I will be very appreciate if u can answer my question. :)

supermerill, 2020-12-01:
sla can only export sl1.
I may add a png export if someone ask for.

ogiwolf, 2020-12-01:
> sla can only export sl1.
> I may add a png export if someone ask for.

Thanks for the fast reply :)
Just wonder why no one export other format such as G-Code for SLA printer.

neophyl, 2020-12-02:
I don't think most SLA printers use gcode.  As the common types use LCD screens to mask off the UV lightsource each layer is essentially a picture.  Of course there's some stuff in the files for how thick each layer is (z axis step) and how long to illuminate each layer but that's about it.

ogiwolf, 2020-12-02:
> I don't think most SLA printers use gcode. As the common types use LCD screens to mask off the UV lightsource each layer is essentially a picture. Of course there's some stuff in the files for how thick each layer is (z axis step) and how long to illuminate each layer but that's about it.

Thanks for telling me. 
And sorry for the stupid question, i am new in 3D printing.
so is there any other type of format for the SLA printer?
such as .photon?

neophyl, 2020-12-02:
There are very few stupid questions :)  
I'm not an expert on SLA printers so I don't know what all the myriad of manufacturers use but I suspect its a bit fragmented with their being no standard at the moment.  It could also be some of the formats the printer manufacturers are using are proprietary.  I'd expect over time for something to emerge as a common type though. 
Even gcode has multiple flavours although that tends to be more documented.

supermerill, 2020-12-02:
In fact, we're speaking here about dlp, as the prusa sl1 and photon printer are dlp.
But prusa market their dpl printer as 'sla', hence some problem when speaking about it.

You're right that sla need some kind of gcode (without extrusion), to move the laser. Currently, there is no sla/sls mode in superslicer. Someone asked for it (for a form2) but I don't really have the time for it, as I prioritize currently the code rework and the integration of belted printers.

neophyl, 2020-12-02:
I'll rejoice when support for belted printers is available.  I had considered getting one but the fact I'd have to use Cura was really putting me off.
SuSie support will be just fabulous.

ogiwolf, 2020-12-03:
> In fact, we're speaking here about dlp, as the prusa sl1 and photon printer are dlp.
> But prusa market their dpl printer as 'sla', hence some problem when speaking about it.
> 
> You're right that sla need some kind of gcode (without extrusion), to move the laser. Currently, there is no sla/sls mode in superslicer. Someone asked for it (for a form1 #227) but I don't really have the time for it, as I prioritize currently the code rework and the integration of belted printers.

Thanks for telling me that the prusa's printer is not the real SLA printer.
I thought it was real SLA printer all the time = =
So what kind of software should i found for the 'real' SLA printer?

closed 2021-06-04 by supermerill