supermerill/SuperSlicer #743: Feature Request/Issue: Concentric Infill (filled) adjustable gap fill amount

Zunzulla, 2020-12-02:
### Version
2.2.53
### Operating system type + version
Windows 10 Pro 10.0.19041 Build 19041

### 3D printer brand / version + firmware version (if known)
Original Prusa Mk3S 3.9.2-3524

When slicing an oval object the gap fill is misplaced (issue #719 already fixed). There is also a gap fill happening in the center, trying to cover the holes that are created due to the oval shape. Could there be a feature, that provides the ability to change the amount of filament extruded into these holes in the center? The misplaced gapfill (if some day placed correctly hehe) works perfectly and fits well. The center fill also works well, especially on cubical and circular objects. But on these oval shapes it's not enough to cover up the holes on the sides. of the center (marked on the picture).

#### Project File (.3MF) where problem occurs
[GapFillOval.zip](https://github.com/supermerill/SuperSlicer/files/5631312/GapFillOval.zip)
[image]

supermerill, 2020-12-02:
not enough plastic?

Zunzulla, 2020-12-02:
> not enough plastic?

It feels like, yes, but this only happens on oval forms

Zunzulla, 2020-12-02:
In addition, the gap fill does not go far enough to the left and right to cover all holes. And the gap fill to fill the seam line is really misplaced, this also occurs in 2.3.55

supermerill, 2020-12-03:
I'll try to incorporate that for the next time i'll work on gap fill, trying to merge all algo in one and merging it in a better way with the perimeter lines.