supermerill/SuperSlicer #744: Adaptive Infill does not work

SAinc, 2020-12-02:
### Version
2.3.55.1
### Operating system type + version
Win 10 19042.662 x64

### 3D printer brand / version + firmware version (if known)
My own CoreXY design running on a Duet 2 with RRF 3.2b4

### Behavior
The adaptive cubic and adaptive support infill settings do not produce any infill. I have tested this with my own profile, and with Prusa's profile with the same results. Testing was done with a 50mm box, generated from SS context menu. I have confirmed that this works on PS 2.3b4.

#### Project File (.3MF) where problem occurs

[Shape-Box.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5631493/Shape-Box.3mf.zip)
(rename to .3mf)

xTremePower, 2020-12-04:
same problem here, it produce a totally infill free object

frezap, 2021-01-13:
Still doesn't seem to work correctly in this version: https://github.com/supermerill/SuperSlicer/actions/runs/475680459

It does do infill, but it doesn't do the adaptive part.

CCS86, 2021-01-26:
I find the same. Both adaptive and support cubic infill just print at a fixed density.

CCS86, 2021-01-26:
Appears to be broken in the PrusaSlicer 2.3.0 official as well. No attempt to densify under this internal bridge:

[image]

VanessaE, 2025-06-16:
I was trying to use this infill type today, and at first it seemed like it wasn't working for me  -- until I turned the density up, similar to @CCS86's previous comment.

I tried slicing a simple 20 mm cube and couldn't get the adaptive behavior until I turned the density up to 42%.  

So ultimately, it's not that this infill type isn't working, but rather it's unclear how it calculates the different levels of density, and what a given percentage setting actually *means*.  The density setting has no clear correlation with...anything obvious, and the size of the part has an effect on the pattern too, and I would not expect that at all.

For example, here's the middle-most layer of a 50 mm cube sliced at 75% density:

[image]

Here's the middle layer of a 200 mm cube sliced at 75% density:

[image]

Er there's no way either of those is 75%-dense at their outer edges (even considering overlapping lines), and that second one kinda fell over, too...

This may need to be re-opened.