supermerill/SuperSlicer #745: Only one perimeter for top layer breaks perimiters for bottom layer

khumarahn, 2020-12-02:
Hi.

I got strange perimeters on a bottom layer with cutout letters:
[image]

Trying to fix this I player with the settings, but none made any difference except printing only one perimeter for the top layer (which I think should have no effect on the bottom layer). With that option unselected (but everything else the same) I get good perimeters:
[image]

I am not attaching the project, sorry, I cannot share that file. I'll see if I can reproduce the same on a different model.

khumarahn, 2020-12-03:
ok, I did not have too look too far. The same happens on the Voron calibration cube, and not only on the bottom layer. And now I can attach the project.
[image]
[Voron_Design_Cube_v7.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5632807/Voron_Design_Cube_v7.3mf.zip)

I get this behavior with appimages generated by github actions for a few recent commits.

supermerill, 2020-12-03:
regression because of #733

VanessaE, 2021-01-18:
(Since there are duplicates of this issue, I have to guess that THIS one is the one to comment on)

I am at commit b6d3143c868c4713f71c20e64d4cc631c93e8400 (git HEAD as of this comment).

I think part of the problem people are having is due to the wording of the "Minimum Width" tooltip, which isn't entirely clear, apart from it being relative to the width of a perimeter.  The default value is 200% which would seem to suggest 2 lines' worth of exposure, but this isn't how it's behaving in reality.

For example, consider this slice of an object I'm working on, at 7.7 mm up (in a 47.6 mm tall part), with my "max quality, strong" profile, which has a 0.1 mm layer height and uses 4 perimeters at 0.65 mm width apiece:

[image]

In this image, I have the Minimum Width set to 900%, because I assumed 4 perimeters, times 2 since they're loops, plus 1 to allow for possible gap fill, (4×2+1)×100%.  As you can see, part of the layer (right) is drawn with up to 4 perimeters plus infill and gap fill, but part of it (left) is drawn with one perimeter plus infill, even though there's room for two perimeter loops plus gap fill, and this island will be almost entirely buried under the one that'll be above it.  

On the other hand, here's a bit higher up on the same object, with the same profile and settings:

[image]

This is exactly how I'd expect the part to be drawn in any case.  Now, if I turn the Minimum Width down a bit, to say 700%, it starts to treat part of the island as a top layer:

[image]

As you can see by the shape of the object, only the tiniest fraction of this layer's outer perimeter would be visible under the next layer above it, due to the slope on the left (the right and the two holes are mild overhangs, the front and back sides are perfectly vertical).  Ditto for the other section of the model mentioned above.

So my assumption seems to be correct -- this Minimum Width threshold is being used not to declare "top layer" status, but unintentionally to declare "enough room", i.e. what's really happening is an all-or-nothing issue relative to whether enough perimeters can fit in the island, regardless if the island is sufficiently-exposed under the next layer above it.

It takes a Minimum Width value of at least **1800%** (yes, eighteen hundred) to get the above section at 7.7 mm (and some portion above and below it) to slice properly, but then other parts that one would expect to be treated as top layers aren't; there is no value I can set that ensures that the entire model is sliced how I'd expect, short of turning the feature off entirely.

What should be happening is it that some portion of the perimeter's *collective* width should need to be exposed under the layer above it, relative to how many perimeters the island could fit, before treating it as a top layer.  A value of 100% would mean that all of the configured perimeters could be exposed under the next layer above, and this would be a good default, as it would ensure that no infill in an area gets exposed without also drawing that area with only one perimeter.

All of the above aside, I can see an argument for making this all-or-nothing behavior an official feature separate from and unrelated to the "one perimeter top layer" feature, as it could provide better results in some cases (such as small, high-details models).  That's provided the "one perimeter top layer" feature is properly fixed. :smile:

Also as another aside, English doesn't usually assign gender to inanimate objects, other than vehicles on occasion.  So instead of "his" in the tooltip, use "its".

Model shown above: [B'struder arm M6, for 11 mm plain gear.stl.zip](https://github.com/supermerill/SuperSlicer/files/5828215/B.struder.arm.M6.for.11.mm.plain.gear.stl.zip)  (and yes, the model is manifold)

Stephan3, 2021-01-24:
In lastest Superslicer increasing minimum widthvalue does not work anymore. even bigger parts still allready have 4 outerwalls on top some stil stay at 1 perimeter.
[image]

I just noticed that they arent printed in toplayerspeed. So the speed setting does not trigger thats there is a toplayer but the perimetercount does.

supermerill, 2021-05-27:
@Stephan3 can you post your project file?

@VanessaE typo fixed!
The algorithm isn't long but it very complicated to get a grasp on.

closed 2021-06-04 by supermerill