supermerill/SuperSlicer #747: Skirt distance from object depending on brim

TheBoojah, 2020-12-03:
This is a feature request.

###Version
2.2.52

### Issue
Today if the skirt distance from object is less than the brim width, the brim will be printed into/over the skirt.
I often use the brim toggle on the plater tab, but do not want the skirt to be further from the object than necessary when not using brim so the skirt is set to 1mm, but when adding brim it will interfere. So I'd have to go into settings and adjust the skirt distance.

### Suggestion
If the skirt distance from object is less than brim width: Add brim width to skirt distance.
Then if skirt distance is 0 it will add a "skirt-brim" to the object if no brim, but also add skirt-brim to the outside of a brim if it is enabled. And if >0 it will get out of the way of the brim, but still be as close as requested to the part if there is no brim.

supermerill, 2020-12-03:
But some people want the skirt to be fuse in the brim, to be able to create a good shield around the object.
Maybe i can add a switch...

TheBoojah, 2020-12-03:
Ah I see. Maybe allow negative distance if brim is active? 0 distance should leave it attached at least.

neophyl, 2020-12-03:
Some also use a skirt with a very high layer count as a draft shield so they want it close to the part and considerably inside the external of the brim. The only problem with the current implementation is when people don’t actually look at the settings they are using.

TheBoojah, 2020-12-04:
If the use case is to attach the skirt to the brim, it does not always work anyway:
[image]
So what is actually needed for the draft shield use case is its own separate brim.

My admittedly small issue with the current implementation is simply that I have to manually fuzz around with manually input numbers to move the skirt around instead of just using a toggle.

cbc02009, 2021-02-01:
I would like to see this feature implemented as well.