supermerill/SuperSlicer #749: Slicing seems to take, literally, forever

lovesegfault, 2020-12-03:
### Version
`b4e4b09a035eb35c1521b14e780b6eb94ba862de`

### Operating system type + version
NixOS Unstable

### Behavior
Slicing the linked project seems to never end. I clicked slice and went to eat breakfast, had some meetings, had lunch, watered my plants and it's still slicing.

This is on my reasonably beefy laptop (6-core Xeon, 32GB ram, nvme ssd).

#### Project File (.3MF) where problem occurs
[CubeGears5-6.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5638928/CubeGears5-6.3mf.zip)

lovesegfault, 2020-12-03:
So, I seem to be unable to reproduce this. I just reopened superslicer and it seemed to slice just fine now. Not sure what happened

supermerill, 2020-12-05:
I hope it's not a bug that can only be reproduced by luck.

closed 2020-12-05 by supermerill