supermerill/SuperSlicer #750: FEATURE REQUEST: Only one perimeter on Bottom surfaces

LazaroFilm, 2020-12-03:
I would love to see an option similar to the already existing `Only one perimeter on Top surfaces:`, but for the bottom layer.
This would be really useful for prints that have letters on the bottom layer (like Benchy) and parts that need to look symmetrical between the top and bottom layers. 
Thanks for your consideration.

MarcPot, 2020-12-04:
However I love the `Only one perimeter on Top surfaces:` function, it has a real reason, your model can have like 20 top surfaces, a model only has 1 bottom surface though.
I don't see the use of extra clutter in the UI (It's quite full already) to add an automated way of just doing a `Height Modifier` and setting the perimeters to `1`.

[image]
[image]

supermerill, 2020-12-04:
> What I would like to see though (And basically fixes what you want) is a preset modifier tab that sets up an entire modifier that you have preset before. (With the option of math added so that you can say 1*{layerheight} instead of 0.2)

sort of duplicate of #412

yeah, a tab for overrides for first layer, and maybe a script window to add more complex ones.

LazaroFilm, 2020-12-06:
I'll mark this as closed since there is a solution for it. Looking forward to see a tab for overrides in the future.

closed 2020-12-06 by LazaroFilm

bigsubgit, 2021-05-30:
Hi everyone, I setup all as indicated, I obtained one perimeter on the bottom, but now it makes only one bottom layer instead of 5 that I want. Is there a way to solve?

MarcPot, 2021-05-30:
Yea it’s a bug, everything after a height modifier won’t be considered bottom layer.

bigsubgit, 2021-05-30:
> 
> 
> Yea it’s a bug, everything after a height modifier won’t be considered bottom layer.

Many thanks.

pranas, 2021-07-01:
The current solution is rather cumbersome. You have to do a lot of steps for every single part. Global modifiers would help but I wish there was an option in print settings, just like there's one for 1 perimeter on top. I think I would just keep it in my profile and use it most of the time since I prefer the finish with 1 perimeter. @supermerill any chance you would reconsider?