supermerill/SuperSlicer #751: Wipe tower line width is 1mm but lines are too dense

khumarahn, 2020-12-04:
Hi Merill,

I appreciate your work on this slicer.

Here is my report on another bug:
[image]

Lines for the wipe tower brim are very dense and very wide. This creates quite a mess on the build plate.

[dual_extruder_calibration_mini_-_extr0.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5639782/dual_extruder_calibration_mini_-_extr0.3mf.zip)

hagan66, 2021-06-01:
I also find this to be a problem. Regular brim settings would be nice for the wipe tower.

closed 2021-06-13 by supermerill

Durahl, 2022-03-27:
Could this possibly have found its way back into 2.4.58.0? 🤔

I'm currently doing some Tri-Color Prints with some of the colors ( white and red ) using very low `Pigment Percentage: 0.2` ( in case it matters ) and found the Wipe Tower Brim to extrude **excessive** amounts of filament to the point where subsequent passes get obstructed by already laid down Filament resulting in the Extruder to grind the Filament to a pulp stopping further Extrusion altogether.

supermerill, 2022-03-28:
#2501