supermerill/SuperSlicer #752: [FEATURE REQUEST] Gradual increase in fan speed

ardyesp, 2020-12-04:
### Description
Soliciting the community to see if there is traction for a feature like this - 
Currently there is an option to disable the parts cooling fan for the first n number of layers, after which the fan starts at default speed. This enhancement request is to gradually increase the fan speed with the layers. It would be helpful for filaments like PLA where fan runs at 100% during the print. This feature would stop the sudden nozzle cooling and allow PID to catchup.

For e.g: I am asking for a configuration option where we can specify a ```fan_default_speed_at_layer``` number. Lets say this number is 5, then fan would be turned on as follows: 
* layer 1: 0%
* layer 2: 20%
* ..
* layer 5: 100% (default speed)

supermerill, 2020-12-07:
Done by prusa, merged with beta-2

ardyesp, 2020-12-08:
Excellent. Thanks for all the work you are putting into the slicer.

closed 2021-06-04 by supermerill