supermerill/SuperSlicer #754: [bug] Zombie slicing with large object

bdbell, 2020-12-04:
### Version
2.2.53

### Operating system type + version
macOS 10.15.7

### 3D printer brand / version + firmware version (if known)
Voron 2.4 running Klipper 0.9.0

### Behavior
Attempting to slice a large STL file (330x330x134).
If slicing at 0.3 layer height, outer corners appear to not slice / render properly.
If slicing at 0.2 layer height, a large quantity of objects appear to fail to slice / render above a certain Z.  They look like zombie objects (no skin).
The object is not complex, just large.
Expectation is just for a standard object slice.

Object in slicer
[image]

Sliced with 0.3 layer height
[image]

Sliced with 0.2 layer height
[image]

#### Project File (.3MF) where problem occurs
Can provide the 3MF over private channels.

supermerill, 2020-12-04:
The gcode viewer seems to have a limit in term of the number of triangles. maybe that's that.

liftbag, 2020-12-05:
> The new gcode viewer seems to have a limit in term of the number of triangles. maybe that's that. There should be a warning dialog anyway.

I don't think that's the reason. I just sliced a large lithophane, full of detail (about 4 minutes to generate the slice on a high-end iMac, the STL file weighs about 300 MB), with no apparent problems.

supermerill, 2020-12-05:
@bdbell I'll need your project file, can you send me a link in discord, or reddit?

closed 2021-08-22 by supermerill