supermerill/SuperSlicer #755: Question: flow ratio %

tecnopolis-ve, 2020-12-04:
### Version
_2.2.53_

### Operating system type + version
_Ubuntu 20.04 LTS_

Hello I've a question, in print settings -> with and flow we can find this options;

[image]

My question is, these flow ratio (100% by default) is calculated over the extrusion multiplier? i.e. if I have 0.9 for extrusion multiplier and 100% flow ratio for bridge it does mean that the extrusion multiplier for bridges is 0.9?

Any help is appreciated.

wildjokerss, 2020-12-04:
Hi 

Not 100% sure but I've tested this out by setting the extrusion multiplier to 0.5 and the line width went from 0.4 to 0.2
I then set the first layer flow to 50% and it went to 0.1. 
It seems to be the percentage of what you have at 100%
if 100% flow means 0.4 width, then 50% flow means 0.2 width. 
From what I've tested, this is the behavior

supermerill, 2020-12-05:
it's multiplicative, as expected.
it's not an override.

closed 2020-12-07 by tecnopolis-ve