supermerill/SuperSlicer #756: Incorrect tooltip or temperature control not working

ardyesp, 2020-12-05:
### Version
v2.3.55_

### Operating system type + version
Windows 64

### 3D printer brand / version + firmware version (if known)
Custom

### Behavior
The tooltip for filament- other layers temperature says setting it to 0 will disable temperature control commands in output gcode. However setting this to zero inserts a 
``` M104 S0 ; set temperature``` 
gcode command at layer# 2.

[image]

liftbag, 2020-12-05:
Isn't this the expected behavior?

ardyesp, 2020-12-05:
> 
> Isn't this the expected behavior?

No, because **disable** would mean the line ```M104 S0``` is not output into the gcode file.

liftbag, 2020-12-06:
> > Isn't this the expected behavior?
> 
> No, because **disable** would mean the line `M104 S0` is not output into the gcode file.

True. You are right.

closed 2021-06-04 by supermerill