supermerill/SuperSlicer #757: Feature Request: Preference to cahnge Settings Folder Location

SigmaRelief, 2020-12-05:
### Version
N/A
### Operating system type + version
N/A
### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
Settings files are currently saved in a pre-defined location which makes it more complicated to share and synchronize across multiple computers.  A simple preference dialoge box option for where to look for settings (either individually by type or as a whole) would greatly simplify this.  There are dozens of workarounds for this, but each is clunky and has drawbacks.

As always, thanks for the amazing work.  This is way less important than the 2.3 merge, but has been bothering me for a year and I would love to hear your thoughts.

MarcPot, 2020-12-05:
That’s already possible (in windows anyway). Make a shortcut for SuperSlicer go to the properties of that shortcut and add `-datadir  “C:/PathToTheFolderYouWant/“` after the target field.

supermerill, 2020-12-05:
eventually, i can search&use for a system variable if it's more convenient.

SigmaRelief, 2020-12-10:
Thanks for that suggestion, it is one I tinkered with, but it doesn't play nicely with launching from the console and portable automation scripts.

Just in case anyone stumbles on this post, in the Windows world, the command would use backslashes without a trailing "\"

ie it would be something like `"C:\Folder Name\superslicer.exe" -datadir "C:\Data Folder"`

closed 2021-06-04 by supermerill

JohnGalt1717, 2022-04-08:
For anyone that discovers this like me there's an easy way to solve this permanently. From a command prompt (not PowerShell, different command there) run this:

mklink /J %APPDATA%\Roaming\SuperSlicer <TargetLocationInQuotesIfSpaces>

This will create a hard link in windows and superslicer is none-the-wiser. I have mine pointed to my onedrive sync so that I never lose updates.