supermerill/SuperSlicer #758: Object Outside Toolpath Error - Filament Flow Calibration

dmpmassive, 2020-12-05:
### Version
Version 2.3.55.2

### Operating system type + version
Windows 10 Pro 64bit Ver 2004 OS Build 19041.630

### 3D printer brand / version + firmware version (if known)
Modified CR-10 S4 runing Duet Boards and RepRap firmware version 3.1.1 Stable

### Behavior
1. Calibration menu >> Filament Flow Calibration
2. Click the "Generate 10% intervals..." Button
3. Generates the five test objects and slices, but it creates a duplicate centred at 0,0,0 and throws an error "Toolpath outside the print area" 

[image]

4. Switching back to the 3d Editor view, the duplicate object at 0,0,0 vanishes. (See attached)

[image]

#### Project File (.3MF) where problem occurs
Not Applicable

supermerill, 2020-12-05:
can i have your project file?

dmpmassive, 2020-12-06:
Of course!  Attached.
[filament_flow_test.zip](https://github.com/supermerill/SuperSlicer/files/5648269/filament_flow_test.zip)

supermerill, 2020-12-08:
Found an other bug, maybe that's that or maybe not, but with the fix it's okay.

closed 2021-06-04 by supermerill