supermerill/SuperSlicer #759: [Bug OR Feature Request] "Enable fan if print layer time is below:"

wildjokerss, 2020-12-05:
### Version
2.3.55.3

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Marlin

### Behavior
* _Describe the problem_ The "Enable fan if print layer time is below:" setting is not applying properly (from my POV)
* _Steps needed to reproduce the problem_
> make a cube, something small, with 1 perimeter, where the layer time is e.g below 4 (mine is 2 seconds)
> Set the "Enable fan if print layer time is below:" to 4
> External perimeter fan is set to 15% **and** Max fan speed for short layer time set to 100%
> Since the external perimeter is set to 15%, even though the layer time is below 4, the fan is not kicking in as per "short layer time" parameters. I think it should since this causes a one wall print to melt.

* _Expected Results_ The "Short layer time" Fan speed to override the "External perimeter fan speed" set above.
* _Actual Results_ "External perimeter fan speed" is taking precedence over "short layer time" parameters.
  * _Screenshots from __*SuperSlicer*__ preview are preferred_
[image]
[image]
[image]

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_
[Shape-Box.zip](https://github.com/supermerill/SuperSlicer/files/5647915/Shape-Box.zip)

supermerill, 2020-12-05:
The external perimeter fan speed is an override, to be sure to not over-cool it. But in your use-case, it's not wanted, so you should disable it.

wildjokerss, 2020-12-06:
Well I don't think that disabling is the answer. I love External perimeter fan speed for the sole purpose of providing better looking external perimeters, however, **if** the layer time is so low, it need to kick in the short layer time parameter.
In this use-case (that isn't much in this specific, I just use this method to calibrate my filaments) I'd have to change the external perimeter setting to compensate for the low layer time.
having it this way, I believe that "Enable fan if print layer time is below:" should override the upper parameters, or at least having a setting that would allow it both ways, so everyone can have it based on what they prefer :P

supermerill, 2020-12-08:
I agree but it's not possible with the current gcode writer, i don't have the "there is only external perimeter" information anymore. Maybe after the rework.

wildjokerss, 2020-12-08:
that's fine, thank you!

FishFPV, 2021-02-14:
How do I get the fan to stay on 100% after the first layer?

supermerill, 2021-06-03:
> How do I get the fan to stay on 100% after the first layer?

tick `run fan at default speed when possible` and set `default fan speed` to 100%

closed 2021-06-03 by supermerill

corrado-c, 2022-02-13:
sorry if I add the comment here, but I noticed in several prints that this function related to the SMALL PERIMETER would be useful, not at the time of the whole layer, am I bullshit?
for example on a time layer of 30 seconds there could be small details that would need to be cooled in order not to collapse.
could help FAN FOR SMALL PERIMETER.

[image]

supermerill, 2022-02-13:
@corrado-c please open a new issue.