supermerill/SuperSlicer #760: Support flow is applied incorrectly at the first layer

kursatu, 2020-12-06:
### Version
_Version of SupserSlicer used goes here, Use `About->About SuperSlicer`_
Built from the github sources.

### Behavior
* _Describe the problem_
I have a lower flow ratio for the support material extrusion width, so that I can break them easily.
But this causes adhesion problems in the first layer.
The better behavior is to use the first layer extrusion width.
* _Screenshots from __*SuperSlicer*__ preview are preferred_
[image]
#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_
[Defect18.zip](https://github.com/supermerill/SuperSlicer/files/5648175/Defect18.zip)

kursatu, 2020-12-06:
The "solid support layer" can also use the "first layers infill speed"
which would be very beneficial.
Otherwise it takes forever to print the first solid support layer, 
[image]

supermerill, 2022-02-03:
Seems like the first layer of the support use the first layer width, in the crurent stable.

For the speed... a bit complicated.