supermerill/SuperSlicer #762: Non-top infills are slightly overextruded with no settings available

belotv, 2020-12-06:
### Version
2.3.55.2

### Operating system type + version
MacOS Big Sur 11.0.1

### 3D printer brand / version + firmware version (if known)
Anycubic Mega i3 - Klipper

### Behavior
* _Describe the problem_
Small infill are usually slightly over extruded (even with a very well calibrated printer, due to the large amount of overlaps on a very small surface) and it contributes to the hull line issues on Benchy (https://help.prusa3d.com/en/article/the-benchy-hull-line_124745) or on the xyzCalibration cube (other main factor being the time to print the layer). 
It can also cause jamming is the printer is printing a lot of small infills on the same layer (especially on Bowden systems with retractation / pressure advance). 

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
Print 3DBenchy/xyzCalibration curve and observe the quantity of plastic deposited on the infill (to support the X for example).
* _Expected Results_
It should be possible to control either the flow, either the gaps between lines.
* _Actual Results_
The only existing parameters are:
- Infill / Solid Infill / Top solid infill Extrusion width
- Top fill flow ratio
- Infill/perimeters overlap
- Anchor solid infill by X mm
Basically, none of these parameters allows a reduction of flow for non-top infills. Changing the extrusion width will make the lines larger/thiner, but additional lines will be created, so the total number of plastic extruded is identical.
Other parameters only control perimeter-infill overlap. It should be possible to control the flow ratio for non-top infills (or line spacing).

  * _Screenshots from __*SuperSlicer*__ preview are preferred_
Mainly has an impact for small infills like:
[image]

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_
[xyzCalibration_cube.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5648541/xyzCalibration_cube.3mf.zip)

supermerill, 2020-12-08:
If you have `enforce 100% fill volume` and all perimeter overlap are at 100% (not the infill/peri), then the printer (on a full solid alyer) will extrude exactly the volume to fill the square, minus the "corners" of the external perimeter
[image]

belotv, 2020-12-08:
I have been rethinking the issue, and I think the root cause is actually a bit different than my initial thought. E-steps are calibrated with a caliper, and enforce 100% fill volume is checked.
However, these very small zig-zags put a huge amount of effort on the filament with pressure advance and it causes some grinding (whatever the tension on the extruder is), and it may fail to retract enough. With a bowden system with PA around 0.5 this is not easily fixable.

Is there a way to disable pressure advance for dense layers / solid infill ? PA is mainly useful for perimeters.

supermerill, 2020-12-08:
you can add that in the 'between extrusion change Gcode' with some if
like 
```
{if extrusion_role=~/.*InternalInfill.*/}SET_PRESSURE_ADVANCE ADVANCE=0
{else}SET_PRESSURE_ADVANCE ADVANCE=0.X
{endif}
```

possibilities:
```
"Perimeter"
"ExternalPerimeter"
"OverhangPerimeter"
"InternalInfill"
"SolidInfill"
"TopSolidInfill"
"BridgeInfill"
"ThinWall"
"GapFill"
"Ironing"
"Skirt"
"SupportMaterial"
"SupportMaterialInterface"
"WipeTower"
"Mill"
"Mixed"
```
https://github.com/prusa3d/PrusaSlicer/wiki/Slic3r-Prusa-Edition-Macro-Language

supermerill, 2020-12-08:
some input for @darkspawn91

* `Infill/perimeter overlap` will stretch the infills (sparse & solid) to push them into the most inner perimeter.
* `Anchor solid infill by X mm` 
  * `Default`: Is the amount of extra mm the solid area is allowed to expand into the sparse infill, to be able to support the perimeters & to be well supported by the previous layer. 
  * `bridge` it's to set the distance the bridge area can enter into the not-bridge area.

tooltips are not enough? where can i improve them?
try with high/low value, it's easy to spot the differences.

darkspawn91, 2020-12-08:
Sorry @supermerill, and thanks again for reply.
Is buldge created by a difference in layer time?
Is not an over extrusion created by solid infill overlap?
I use “0” on “infill overlap” and “0.5mm” on “default anchor solid infill by X mm” (as someone told in another post), but nothing has changed. 
I’ll try to set it also on 0.
Can also have “enforce full fil volume” set to “100%” an impact on infill over extrusion?

Anyway you do your best, thanks for your work.

supermerill, 2020-12-08:
> Can also have “enforce full fill volume” set to “100%” an impact on infill over extrusion?

it's a switch, so you can activate or deactivate, there is no '%'.

darkspawn91, 2020-12-08:
Yeah, it’s a switch. Sorry. 
I’ll try to change also this setting. 
Thanks

closed 2021-06-04 by supermerill