supermerill/SuperSlicer #764: Bridge acceleration is used for overhang perimeter below the bridge threshold causing bulging

belotv, 2020-12-06:
### Version
2.3.55.2

### Operating system type + version
MacOS 11.0.1

### 3D printer brand / version + firmware version (if known)
Anycubic Mega i3 - Klipper

### Behavior
* _Describe the problem_
I use very slow acceleration for bridging (250), to ensure proper binding, and normal acceleration for perimeters (3500). There are some overhang perimeter on 3D Benchy model. These overhangs are below the threshold for bridge flow, so normal settings should be used (only overhang speed shall be used).
* _Expected Results_
Default acceleration is used for these overhangs. If this is expected behaviour, then we should have the possibility to give an overhang acceleration control speed because this causes a lot of disgraceful bulging on the perimeter.
* _Actual Results_
Bridge acceleration is used for these overhangs (but speed and flow are ok).
  * _Screenshots from __*SuperSlicer*__ preview are preferred_
[image]

#### Project File (.3MF) where problem occurs
_Upload a SuperSlicer Project File (.3MF) (`File -> Save` / `Save Project`)_
[benchy.gcode.zip](https://github.com/supermerill/SuperSlicer/files/5648863/benchy.gcode.zip)
[benchy.3mf.zip](https://github.com/supermerill/SuperSlicer/files/5648865/benchy.3mf.zip)

supermerill, 2020-12-08:
I think the problem lies on problems on the slicing side, as it shouldn't have that erratic polygons.

belotv, 2020-12-08:
Thanks for the remark, erratic polygons are due to the curve smoothing function. I just disabled it and they disappeared. However it does not really fix the issue.
I would like the hull to be printed slower (with bridge settings), but i do not want the bridge acceleration to be used, only the speed and fan. This does not seems to be feasible today.

supermerill, 2020-12-08:
curve smoothing fixed.

So keeping external acceleration until the switch to bridge flow?
* overhangs detected : use bridge fan & overhang speed
* bridge flow threshold: use bridge flow & bridge acceleration

robthide37, 2020-12-09:
i think it makes sense to bind the accel with the flow change

belotv, 2020-12-09:
Thanks for your replies. I do not really agree that it make sense there. Contrarily to the other features, overhangs are extension of perimeters, which do not requires the same binding specificity as bridges. Here is an example (top view):

[image]

closed 2021-06-04 by supermerill