supermerill/SuperSlicer #765: Slicer crash, looking for nightly but can't find them

ProfEngr, 2020-12-06:
It's crashing when I click slice in 2.3.55.2 and 2.2.54.2 using this profile and this file...

[files.zip](https://github.com/supermerill/SuperSlicer/files/5649535/files.zip)

Other similar issues suggest grabbing the nightly, but I've been unable to figure out how. The artifacts referenced on the code page are expired. I know I missed something simple. I didn't have any luck the last time I tried to compile the source for win64.

ETE-Design, 2020-12-06:
You can find "Nightly" here: https://github.com/supermerill/SuperSlicer/actions

ProfEngr, 2020-12-06:
I've been all over that link, but the d/l isn't jumping out at me.

ETE-Design, 2020-12-07:
@ProfEngr https://github.com/supermerill/SuperSlicer/commit/8f7e8322a9027ecf3a108309b4c6ab462d66f257

ProfEngr, 2020-12-07:
@ETE-Design I got an email reply with a link that got me where I needed to go. I don't know why I couldn't get there with the one above. Thanks! Now to figure out the new printer profile structure due to the way they handle the octoprint API.

That nightly doesn't crash when slicing the 2-wall extrusion multiplier calibration cube.

closed 2020-12-07 by ProfEngr

ProfEngr, 2020-12-07:
OK, so the new physical printer "layer" in the hierarchy stores the API/IP but doesn't duplicate the actual printer profile. It's just attached to it in a way that's visually confusing. Beginning to make sense. Choose the physical printer on the preview window to add export to printer. The preset itself only exports to file.