supermerill/SuperSlicer #766: Import Config Bundle from 2.2.5.4

thierryzoller, 2020-12-07:
### Version
2.3.55.2

Importing a config bundle that I exported from 2.2.5.4 results in the following error:
ConfigBase::set_deserialize() failed for parameter "solid_fill_pattern", value "monotonous". No settings are loaded.

supermerill, 2020-12-08:
Can i have the config bundle?

thierryzoller, 2020-12-08:
[Uploading SuperSlicer_config_bundle.zip…]()

thierryzoller, 2020-12-08:
[SuperSlicer_config_bundle.zip](https://github.com/supermerill/SuperSlicer/files/5658931/SuperSlicer_config_bundle.zip)

supermerill, 2020-12-08:
Can't reproduce.
You used this menu entry?
[image]

thierryzoller, 2020-12-08:
Video 
https://youtu.be/6I8oUeBMTAE

closed 2021-06-04 by supermerill