supermerill/SuperSlicer #769: Rotation not updated in Object manipulation fields.

TheAndr0id, 2020-12-07:
### Version
2.2.53.2

### Operating system type + version
Linux x86_64
Windows 7/x64

### Behavior
If an object is rotated (either with the tool or filling in the fields in the Object manipulation box), the values in the X, Y and Z rotation fields are always 0.  If a value is entered the object is rotated as requested, but the field reverts back to a 0 value.  The undo marker is present, but not the values.

This is an upstream fix as PrusaSlicer (2.2.0) shows the same behavior.

[image]

supermerill, 2020-12-08:
Did they have an issue on it?

TheAndr0id, 2020-12-08:
I did a search for open issues with "rotate" and "rotation" and found nothing that looked to apply.  I'm surprised that this hasn't been reported earlier.

supermerill, 2020-12-08:
I found https://github.com/prusa3d/PrusaSlicer/issues/777
But nothing on keeping the rotation in the box.
Can you create an issue in the prusa github about it, to understand why it's that way?
I can see an advantage: it's easier to do x then y than to try to enter a xyz rotation.

TheAndr0id, 2020-12-08:
After some grepping, I found the following:

In src/slic3r/GUI/GUI_ObjectManipulation.cpp:

```
if (m_world_coordinates) {
      m_new_rotate_label_string = L("Rotate");
      m_new_rotation = Vec3d::Zero();
      m_new_size     = selection.get_scaled_instance_bounding_box().size();
      m_new_scale    = m_new_size.cwiseProduct(selection.get_unscaled_instance_bounding_box().size().cwiseInverse()) * 100.;
    } else {
      m_new_rotation = volume->get_instance_rotation() * (180. / M_PI);
      m_new_size     = volume->get_instance_transformation().get_scaling_factor().cwiseProduct(wxGetApp().model().objects[volume->object_idx()]->raw_mesh_bounding_box().size());
      m_new_scale    = volume->get_instance_scaling_factor() * 100.;
    }
```

If the manipulation box is set to "Global coordinates, then the rotation values are all set the zero via the Vec3d:Zero() call.  If they are set the Local coordinates then they show the actual rotation values as expected.

I then did some extra digging.  This page:

https://help.prusa3d.com/en/article/object-manipulation-panel_1757

Explains what the difference between the Global and Local coordinate systems are and how it relates to rotated items.   This kind of clarifies why the rotation values would always be zero, but the Global/Local system isn't intuitive (to me anyway). 

There is a ticket open requesting an update to separate the object manipulation from the Local/Global space.  We'll see where that goes as this feature was a major item with the 2.2 release...

Personally I think this needs some clarification, but it is what it is.  Closing issue.

closed 2020-12-08 by TheAndr0id