supermerill/SuperSlicer #770: Feature Request: Gap Filling Width Bounds

CCS86, 2020-12-07:
2.2.54 / Win10 x64

I am wondering if the gap filling algorithm could be further improved by applying a bound to the maximum gap filling line width. 

I have found that gap filling can attempt a line width essentially 200% of nozzle width. Here is an example where two full width lines become a single gap filling line, when the part contracts just slightly:

[image]

[image]

IMO, a better result would be achieved with either 2 equal-width lines, slightly thinner than default. Or with one default width line, and another reduced width line.

I realize that because the area is bounded by an outside wall, on both sides, the slicer sees this as 1 perimeter. Once it can't fit that perimeter, it gets kicked to the gap filling algorithm, right? I'm not sure how challenging this would be to implement. But, if you could specify a maximum gap filling line width, and it5 came and filled that area once. Then, it could come back and fill it again with a reduced width line.

qwewer0, 2020-12-07:
I think this is already fixed in the latest PS/SS.

CCS86, 2020-12-07:
> I think this is already fixed in the latest PS/SS.

Thanks, I see that now.

Downloaded 2.3.55.1 and loaded the same part and config, but I am still getting a fat gap fill on layer 132.

[OP10-FixSS.3mf.txt](https://github.com/supermerill/SuperSlicer/files/5655689/OP10-FixSS.3mf.txt)

supermerill, 2020-12-08:
look at the relaese note of 2.3.55.2

qwewer0, 2020-12-08:
> If possible, an extra perimeter is extruded instead of a fat gapfill, to be able to stay below the perimeter width without overextruding.

I had [2.3.55.1](https://github.com/supermerill/SuperSlicer/releases/tag/2.3.55.1) in mind, but if that isn't applied to non loops, then don't mind me.

supermerill, 2020-12-08:
> I had [2.3.55.1](https://github.com/supermerill/SuperSlicer/releases/tag/2.3.55.1) in mind, but if that isn't applied to non loops, then don't mind me.

oh you're right, sorry.

qwewer0, 2020-12-08:
@supermerill Is that applies to this issue too?

supermerill, 2020-12-08:
You're right that it's the 2.3.55.1, not the 2.3.55.2

supermerill, 2022-03-23:
now there is a threshold

CCS86, 2022-03-23:
> now there is a threshold

Is it just a built in limiter, or a new setting?

supermerill, 2022-03-24:
overlapping periemter aren't a on/off anymore, but are a threshold between 0% (off) and 100% (on)
so you can set the point where you don't want gapfill anymore and prefer having perimeters that overlap a bit each other (and so overextrude a bit)

It's not the ultimate solution, but it was a simple improvment, waiting for a better alternative.

CCS86, 2022-03-24:
Ah, so overlaps still aren't flow compensated locally?

supermerill, 2022-03-24:
no

closed 2022-04-01 by supermerill