supermerill/SuperSlicer #772: Bug: Unwanted filament changes triggered when using Hilbert Curve infill

JasonGuthrie, 2020-12-07:
### Version
_Version of SupserSlicer used goes here, Use `About->About SuperSlicer`_
2.3.55.2

### Operating system type + version
W10 - Pro

### 3D printer brand / version + firmware version (if known)
Prusa Mk3 + V3.9.0 (1.0.6)

### Behavior
When using Hilbert Curve infill, mysterious filament changes are requested by the printer. This causes no issue to print quality but requires the machine to be attended to for a needless filament change every 10 minutes or so.

The filament changes are not visible in the slicer preview or the g-code preview, they only show their faces when actually printing.

**Note:** This issue also occurs in **Vanilla Prusa Slicer 2.3.0 - Alpha 3** (not tested anything newer due to general stability issues).

supermerill, 2020-12-08:
project file?

JasonGuthrie, 2020-12-08:
> project file?

No need anymore, the problem has been resolved. 

Turns out the issue was hardware related (filament sensor was triggering during Hilbert Curve infill, the problem went away when I turned the sensor off.

Thanks for your attention though,
Cheers.

closed 2020-12-08 by JasonGuthrie