supermerill/SuperSlicer #773: SuperSlicer 2.3.55.2 segfaults on Monotonic as support material interface pattern

jhoblitt, 2020-12-08:
I'm not sure if those choice of options makes any sense but it does trigger a segfault.

### Version
`2.3.55.2` / `SuperSlicer-ubuntu_18.04-2.3.55.2.AppImage`

### Operating system type + version

```
 ~ $ cat /etc/redhat-release 
Fedora release 32 (Thirty Two)
```

### Behavior
* selecting `Monotonic` as the support material interface pattern and then starting the slice causes a segfault.
```
Segmentation fault (core dumped)
```

#### Project File (.3MF) where problem occurs

[bearing_block.zip](https://github.com/supermerill/SuperSlicer/files/5656021/bearing_block.zip)

https://youtu.be/kP4Y63UvVvw

supermerill, 2020-12-08:
The video show a bearing holder, and you send me the bearing block, and it slice fine, and doesn't have supports.
You are sure it's the right project file?

supermerill, 2020-12-08:
maybe same bug as #763

jhoblitt, 2020-12-08:
Yes, sorry, I must have zipped the wrong file to make github happy.
[bearing_holder.zip](https://github.com/supermerill/SuperSlicer/files/5656473/bearing_holder.zip)

supermerill, 2020-12-08:
Can't reproduce on my dev version, so it's most probably the same bug as 763

jhoblitt, 2020-12-08:
I'm happy to try to confirm.  Is there a nightly appimage or do I need to build from src?

supermerill, 2020-12-08:
in ~two hour, in the actions menu here.

closed 2021-06-26 by supermerill