supermerill/SuperSlicer #774: Custom Default Settings?

CCS86, 2020-12-08:
I'm trying to figure out the best way to manage my print settings.

The general theme is that I keep refining some base print settings, that work best on my machine (line widths, overlaps, acceleration, etc), and with those applied globally, I have different profiles setup for different: layer height, speed, infill, etc.

Currently, any time I improve a "base" setting, I have to manually push it across into all of my profiles. This is very time consuming, and I hope there is a better way. 

Seemingly, the built in "default setting" lock/unlock functionality is perfect for this. It seems like allowing for a toggle between [default / custom default] would allow a clean way for the user to tune their own baseline default settings. Maybe just having a "nuclear option" to nuke all custom defaults, and restore the Superslicer defaults. Or, having that option on a per setting basis.

Thoughts?

supermerill, 2020-12-08:
create you own vendor profile.
take a simple one like the voron.ini (it's in resources/profile), copy it and rename things, change settings, create new printer, etc...

CCS86, 2020-12-08:
I guess if I had solid default values, that could work for me.

But since my defaults are always evolving, it seems even more cumbersome to update them via text editing.

CCS86, 2020-12-08:
Another idea is instead of worrying about "defaults":

 Settings could just have a "link" button. Click to link, then select the profile you want to link to. This could be even more powerful, without disrupting the existing default setting functionality.

supermerill, 2020-12-08:
There is an issue about a "write/erase on vendor profile".
But I don't have time for that in the medium term.

Sebazzz, 2021-05-03:
@supermerill Is it correct that `inherits` only works when it refers to a vendor profile?

supermerill, 2021-05-03:
@Sebazzz
yes, it's to shorten the file, to avoid repeating ad nauseum the same settings on each section.

closed 2021-06-04 by supermerill

CCS86, 2021-06-04:
> Another idea is instead of worrying about "defaults":
> 
> Settings could just have a "link" button. Click to link, then select the profile you want to link to. This could be even more powerful, without disrupting the existing default setting functionality.

No interest in this suggestion?

supermerill, 2021-06-04:
I was mass-cleaning open derelict issues. 

I don't have time this year for revamping profiles bundles anyway.
If something new happens, feel free to open a new issue or re-open this one.