supermerill/SuperSlicer #776: preview of tools are all solid black

johnmeacham, 2020-12-08:
### Version
2.3.55.2

### Operating system type + version
Linux ubuntu 18.04

### 3D printer brand / version + firmware version (if known)
e3d toolchanger

### Behavior
* When looking at the gcode preview showing tool paths, all tools are represented by the color black or randomly another color occasionally.
* slice anything that uses multiple tools and click preview
* The tool path should be rendered in either the tool color specified in the printer specification or the filament color chosen for that tool. (either would make sense)
* tools are rendered black
  * 
[image]

#### Project File (.3MF) where problem occurs

[DualStrusion-spiralCylinder_m1.3mf.gz](https://github.com/supermerill/SuperSlicer/files/5660900/DualStrusion-spiralCylinder_m1.3mf.gz)

supermerill, 2020-12-08:
Can you try on the lastest nightly? (actions menu /\)
[image]

johnmeacham, 2020-12-08:
It appears to be fixed in this build. thanks! Recently migrated from stock slic3r and am really liking it so far. 
 https://github.com/supermerill/SuperSlicer/actions/runs/406989046

closed 2020-12-08 by johnmeacham