supermerill/SuperSlicer #777: Segmentation fault: 11 on osx Big Sur

liftbag, 2020-12-08:
### Version
_2.3.55.3 fix build C/C++ CI macos #194_
_2.3.55.3 fix for naming thread when restricted to one thread. C/C++ CI macos #195_

### Operating system type + version

[image]

### Behavior
The program closes at start with error: _segmentation fault: 11_
The latest working release is the _2.3.55.3 fix error in Voron.ini update ender3 manual mmu C/C++ CI macos #192_

[image]

cknic, 2021-01-01:
I have the same exact issue with the latest as well.

liftbag, 2021-01-14:
The "revert deps missing C/C++ CI macos #199" version still doesn't work.

liftbag, 2021-01-18:
I enclose the Apple crash report, maybe it is useful?

[SuperSlicer segmentation fault 11.zip](https://github.com/supermerill/SuperSlicer/files/5831864/SuperSlicer.segmentation.fault.11.zip)

liftbag, 2021-01-20:
Well, reading the issue #897 (Windows related), I wanted to try to open the program as administrator and the nightly build opened correctly. I don't think it's a file permissions problem. Opening it as administrator, the profiles are ignored and the program must be initialized.
Evidently, there are some settings that are no longer accepted by recent nightly builds. It will be tough, but I'll try to investigate better.

supermerill, 2021-01-23:
The crash report doesn't report anythign that i can fix, as it's in the wxwidget lib that the problem occur.
will merge with latest prusa update for macos.

liftbag, 2021-01-23:
> The crash report doesn't report anythign that i can fix, as it's in the wxwidget lib that the problem occur.
> will merge with latest prusa update for macos.

Ok. So, something has changed with wxwidget from the _fix error in Voron.ini update ender3 manual mmu C/C++ CI macos #192_ version?

issic, 2021-02-09:
Any update on this?

issic, 2021-02-10:
Will this help?
change this in PrusaSlicer-master/deps/wxWidgets/wxWidgets.cmake:
-DwxUSE_LIBPNG=builtin
and rebuild deps and compile again.

and fixed in the latest master branch of PrusaSlicer, you must:
Delete all the previous build artifacts including the dependencies and start from scratch.

gretel, 2021-02-17:
`2.3.55.2` working fine on osx `11.3 Beta (20E5172i)` for me.

liftbag, 2021-02-20:
> `2.3.55.2` working fine on osx `11.3 Beta (20E5172i)` for me.

2.3.55.2 works fine. The issue is related to the latest nightly builds based on 2.3.55.3.

royal32, 2021-02-24:
Hey all,

I've figured out this crash- PrusaSlicer encountered the same thing in Nov. It's mentioned [here](https://github.com/prusa3d/PrusaSlicer/issues/5232), and was caused by a wxWidgets version that is no longer compatible with macOS 11 Big Sur. [This commit](https://github.com/prusa3d/PrusaSlicer/commit/264ce71e94bdd8e59691139dd612ee6ee1a59f99) which forces *every* mac build (not just M1 macs) to use wxWidgets 3.1.4-patched.
I've confirmed this and successfully built SS 2.3.55.5 on Big Sur 11.2.1. 
I've tagged merill on discord so he is aware.

royal32, 2021-03-12:
Prusa has now made 3.1.4-patched the version used to build [on all platforms.](https://github.com/prusa3d/PrusaSlicer/commit/d7a786e8c8d82eb0be95e81094665e0fb23ce66b#diff-f9fad47d306cc2cfccde494f7e3154bd2244685f7fb70f01d73375dc60de9a6b)
@supermerill can we get the new [wxWidgets.cmake](https://github.com/prusa3d/PrusaSlicer/blob/master/deps/wxWidgets/wxWidgets.cmake) merged so that we have mac builds again :)

gretel, 2021-03-12:
that'd be 🥳

LazaroFilm, 2021-03-14:
> Hey all,
>
> I've figured out this crash- PrusaSlicer encountered the same thing in Nov. It's mentioned [here](https://github.com/prusa3d/PrusaSlicer/issues/5232), and was caused by a wxWidgets version [...]

Could you share your build?

royal32, 2021-03-14:
> Could you share your build?

[Here you go](https://github.com/supermerill/SuperSlicer/issues/977#issuecomment-786259606)

gretel, 2021-03-17:
@royal32 thanks! still crashing (on slicing) on `11.3` on a `m1` mac.

```
Thread 10 Crashed:: slic3r_BgSlcPcs
0   libc++.1.dylib                	0x00007fff204b8437 std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::operator=(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 9
1   com.merill.slic3r/            	0x0000000102de7f25 Slic3r::GCodeProcessor::apply_config(Slic3r::PrintConfig const&) + 421
2   com.merill.slic3r/            	0x0000000102e0c15a Slic3r::GCode::_do_export(Slic3r::Print&, __sFILE*, std::__1::function<void (std::__1::vector<Slic3r::ThumbnailData, std::__1::allocator<Slic3r::ThumbnailData> >&, std::__1::vector<Eigen::Matrix<double, 2, 1, 2, 2, 1>, std::__1::allocator<Eigen::Matrix<double, 2, 1, 2, 2, 1> > > const&, bool, bool, bool, bool)>) + 122
3   com.merill.slic3r/            	0x0000000102e0aa26 Slic3r::GCode::do_export(Slic3r::Print*, char const*, Slic3r::GCodeProcessor::Result*, std::__1::function<void (std::__1::vector<Slic3r::ThumbnailData, std::__1::allocator<Slic3r::ThumbnailData> >&, std::__1::vector<Eigen::Matrix<double, 2, 1, 2, 2, 1>, std::__1::allocator<Eigen::Matrix<double, 2, 1, 2, 2, 1> > > const&, bool, bool, bool, bool)>) + 1014
4   com.merill.slic3r/            	0x000000010301be31 Slic3r::Print::export_gcode(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, Slic3r::GCodeProcessor::Result*, std::__1::function<void (std::__1::vector<Slic3r::ThumbnailData, std::__1::allocator<Slic3r::ThumbnailData> >&, std::__1::vector<Eigen::Matrix<double, 2, 1, 2, 2, 1>, std::__1::allocator<Eigen::Matrix<double, 2, 1, 2, 2, 1> > > const&, bool, bool, bool, bool)>) + 657
5   com.merill.slic3r/            	0x0000000103334fba Slic3r::BackgroundSlicingProcess::process_fff() + 426
6   com.merill.slic3r/            	0x0000000103337a20 Slic3r::BackgroundSlicingProcess::thread_proc() + 304
7   com.merill.slic3r/            	0x00000001033393e2 boost::detail::thread_data<Slic3r::BackgroundSlicingProcess::start()::$_1>::run() + 18
8   com.merill.slic3r/            	0x000000010385c118 boost::(anonymous namespace)::thread_proxy(void*) + 120
9   libsystem_pthread.dylib       	0x00007fff2051e954 _pthread_start + 224
10  libsystem_pthread.dylib       	0x00007fff2051a4a7 thread_start + 15
```

supermerill, 2021-03-18:
someone here can try the latest nightly build?
https://github.com/supermerill/SuperSlicer/suites/2281540987/artifacts/47710605https://github.com/supermerill/SuperSlicer/suites/2281540987/artifacts/47710605

royal32, 2021-03-18:
> someone here can try the latest nightly build?
> https://github.com/supermerill/SuperSlicer/suites/2281540987/artifacts/47710605https://github.com/supermerill/SuperSlicer/suites/2281540987/artifacts/47710605

Works great on macOS 11.2.1! :D

gretel, 2021-03-18:
@supermerill thanks; on `11.3` it's still crashing on slicing:

```
Thread 10 Crashed:: slic3r_BgSlcPcs
0   libc++.1.dylib                	0x00007fff204c8437 std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::operator=(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) + 9
1                                 	0x0000000103125dd5 Slic3r::GCodeProcessor::apply_config(Slic3r::PrintConfig const&) + 421
2                                 	0x000000010314a00a Slic3r::GCode::_do_export(Slic3r::Print&, __sFILE*, std::__1::function<void (std::__1::vector<Slic3r::ThumbnailData, std::__1::allocator<Slic3r::ThumbnailData> >&, std::__1::vector<Eigen::Matrix<double, 2, 1, 2, 2, 1>, std::__1::allocator<Eigen::Matrix<double, 2, 1, 2, 2, 1> > > const&, bool, bool, bool, bool)>) + 122
3                                 	0x00000001031488d6 Slic3r::GCode::do_export(Slic3r::Print*, char const*, Slic3r::GCodeProcessor::Result*, std::__1::function<void (std::__1::vector<Slic3r::ThumbnailData, std::__1::allocator<Slic3r::ThumbnailData> >&, std::__1::vector<Eigen::Matrix<double, 2, 1, 2, 2, 1>, std::__1::allocator<Eigen::Matrix<double, 2, 1, 2, 2, 1> > > const&, bool, bool, bool, bool)>) + 1014
4                                 	0x000000010335a3c9 Slic3r::Print::export_gcode(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, Slic3r::GCodeProcessor::Result*, std::__1::function<void (std::__1::vector<Slic3r::ThumbnailData, std::__1::allocator<Slic3r::ThumbnailData> >&, std::__1::vector<Eigen::Matrix<double, 2, 1, 2, 2, 1>, std::__1::allocator<Eigen::Matrix<double, 2, 1, 2, 2, 1> > > const&, bool, bool, bool, bool)>) + 649
5                                 	0x000000010367374a Slic3r::BackgroundSlicingProcess::process_fff() + 426
6                                 	0x00000001036761b0 Slic3r::BackgroundSlicingProcess::thread_proc() + 304
7                                 	0x0000000103677b72 boost::detail::thread_data<Slic3r::BackgroundSlicingProcess::start()::$_1>::run() + 18
8                                 	0x0000000103b9b7b8 boost::(anonymous namespace)::thread_proxy(void*) + 120
9   libsystem_pthread.dylib       	0x00007fff2052e954 _pthread_start + 224
10  libsystem_pthread.dylib       	0x00007fff2052a4a7 thread_start + 15
```

besides, the url you posted is borked :)

supermerill, 2021-03-18:
@gretel i don't see what's wrong where it crashs. Can you share your project file that crashs on slicing?

> besides, the url you posted is borked :)

Was too ctrl+v happy.

gretel, 2021-03-18:
> @gretel i don't see what's wrong where it crashs. Can you share your project file that crashs on slicing? Maybe it's not something macos-related.

sure, here's the `stl`:
[maquet.stl.zip](https://github.com/supermerill/SuperSlicer/files/6164536/maquet.stl.zip)

liftbag, 2021-03-19:
The last nightly __#944 Spiral vase mode should turn off some of SuperSlicer's new settings__ opens without errors. I think we can close the issue.

gretel, 2021-03-19:
@liftbag all the builds referred to in this issue open on big sur; but crash on slicing for me.

gretel, 2021-03-20:
> I need the project file : before clicking on the "clice" button that would makes it crash, please click on file-> save as and zip & paste the file here.

@supermerill there you go:
[maquet.3mf.zip](https://github.com/supermerill/SuperSlicer/files/6174714/maquet.3mf.zip)

closed 2021-06-07 by supermerill

gretel, 2021-06-07:
@supermerill thanks