supermerill/SuperSlicer #778: Feature Request: Pattern Angle for Support Material Interface

pemerick07, 2020-12-09:
### Behavior
I'd like the ability to set the pattern angle for the support material interface in the same way it can be done for the support material and raft. This would be very helpful for printing rafts with dissolvable support for materials that warp significantly such as PPE-PS, PP or PA12. With the material interface lines all going in the same direction, the interface is very weak in the opposite direction which allows the primary material to curl along the edges. Cura supports this feature and reduces the curling significantly, but has other issues that make me not want to use it.

supermerill, 2023-12-28:
I think it's added now.

shengyan, 2024-02-25:
In 2.5.59.8, the first layer of the support interface is not affect by the support_material_interface_angle setting. It always has a 0 degree angle.
[image]

shengyan, 2024-07-02:
Was trying out the "not support material" fix #4333 in the 2.5.59.13, and found an working edge case to this new feature.
[image]
The image shows that if the support interface is rested above a narrow surface, then the interface pattern angle
works as intended, see the "second floor". It is not working though on a wider surface, see the "first floor".

Also the #4333 issue is not fully fixed. There is still unintended support material on the build surface. 
[image]

supermerill, 2024-07-02:
@shengyan can you provide the project file so I can test it?

shengyan, 2024-07-03:
I have done some more testing, and find that

1. it is the 1st most layer causing the problem(it always work on the second floor);
2. it works on the 1st most layer if the interface is reaching to the edge of the previous top infill(object1 reach to two ends, object2 does no reach to any, object3 reach to the side).
3. support material extrusion width does no affecting the 1st layer(object4, not just 1st most layer).
[image]

[project file.zip](https://github.com/user-attachments/files/16081842/body1.zip)

shengyan, 2024-07-03:
There are support material on the build surface, so the 1st most interface layers in object1 and object2 are not really the 1st most layer?
maybe it is relate to some counter in the code?
[image]

supermerill, 2024-07-03:
Yes, I found the problem:
The first support layer, if it's an interface, was confused with a raft.
Fixed that for next nightly.

> support material extrusion width does no affecting the 1st layer(object4, not just 1st most layer).

That's because it's bridging on top of the object, so it use the bridge flow.