supermerill/SuperSlicer #779: Flow calibration prints - 10% intervals -10% print looks exactly the same as 0 print on 2% intervals

brymck1, 2020-12-09:
### Version
2.2.53

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro, SKR mini e3 v1.2

### Behavior
Ensure extrusion multiplier on filament is 1.0
Slice filament flow calibration test with 10% intervals around current value (-20 to +20%)
Note that best print for my printer looks like -10% which is near flawless, 0 looks a little overextruded/blobby
Decide to slice the 2% intervals below current value while still configured at 1.0 to ensure it shouldn't be between 0 and -10%
0 print (multiplier=1.0) on the 2% intervals looks exactly the same as -10% on the 10% intervals (multiplier=0.9)
Every other item (-2%, -4%, -6%, -8%) all look underextruded/gappy
Noted that 2% interval prints are taller/different shapes than the 10% prints

#### Project File (.3MF) where problem occurs
Attached. Let me know if you want the gcode of the slices as well, if there's something funky with my config.
[calibration.zip](https://github.com/supermerill/SuperSlicer/files/5663301/calibration.zip)

supermerill, 2020-12-09:
> Noted that 2% interval prints are taller/different shapes than the 10% prints

They are to amplify a bit the effect of a not optimal flow

brymck1, 2020-12-09:
So in order, I printed:

10% at 1.0
2% at 1.0
2% at 1.0
10% at 1.0

The 0 labeled one from 2% and -10 labeled from 10% looked nearly identical on all.

Then, printed 10% at .9 and the 0 looks great...then printed 2% at .9 and all of them are underextruded including 0.

I'll swap the filament and try again.

supermerill, 2020-12-09:
maybe extruding more mayers for the 2% test are a bad idea, but if you are your bed perfectly levelled, it shouldn't

brymck1, 2020-12-13:
Okay, so...stock hotend was apparently loose on the carriage. I had a Micro Swiss hotend I'd been meaning to upgrade to, so I did. Everything is secure now. Validated with bed leveling test in slicer that it is perfect on corners, and I have a bltouch doing 9 point mesh each print to boot. 

Loaded new filament just in case that was the culprit after all this. Printed bed leveling test to ensure everything was good and it looked as it should. Printed the 10% extrusion test at 100% flow and -20% is cleanest, 0 and -10 are overextruded. Adjusted filament flow to 80%, then sliced -2% steps, and every one is underextruded including the 0% tab. I'm re-printing just to make sure but I can't detect anything wrong with my printer physically at this point.

I  looked at the gcode and it looks like it's supposed to be extruding the same amount on each move for the top layer between the 0% on the .8 10% test, and the 0% on the .8 2% test:

.8 flow multiplier at 10% steps top layer 1.84mm:
first object top layer prints extrusions .00880 to .00935
second object top layer prints extrusions .00990 to .01021
third object top layer prints extrusions .01066 to .01169
fourth object top layer prints extrusions .01173 to .01211
fifth object top layer prints extrusions .01279 to .01321

.8 flow multiplier at 2% steps top layer 3.24mm:
first object top layer prints extrusions .00981 to .01012
second object top layer prints extrusions .01002 to .01035
third object top layer prints extrusions .01024 to .01057
fourth object top layer prints extrusions .01045 to .01079
fifth object top layer prints extrusions .01066 to .01169